FreeOpcUa/opcua-asyncio #1336: Can't write value to the opc server

John202105, 2023-06-15:


John202105, 2023-06-15:
hello, everyone, I had built a opc client with asyncua and i can read the tags from the opc server of s7-1200. but, everytime, when i want to write data to the tag, the opc connect always auto close.  please help me find the problem and give me some advice. thanks .

`nodepath = []
                        nodepath = nodepath+node
                        nodepath.append("5:KOH")
                        tag_fullname = await client.nodes.root.get_child(nodepath)
                        print(tag_fullname)
                        value = await tag_fullname.get_value()
                        print(value)

                        await tag_fullname.set_value(78.5)
                        print('*********************')`

the run info : 

*********************************
2023-06-15 16:17:31,879 - MyLib - INFO - Start Get Data
*********************************
ns=5;i=12
8.5
2023-06-15 16:17:32,190 - asyncua.client.client - INFO - disconnect
2023-06-15 16:17:32,190 - asyncua.client.ua_client.UaClient - INFO - close_session
2023-06-15 16:17:32,510 - asyncua.client.ua_client.UASocketProtocol - INFO - close_secure_channel
2023-06-15 16:17:32,511 - asyncua.client.ua_client.UASocketProtocol - INFO - Request to close socket received
2023-06-15 16:17:32,513 - MyLib - WARNING - Reconnecting in 2 seconds
2023-06-15 16:17:32,516 - asyncua.client.ua_client.UASocketProtocol - INFO - Socket has closed connection
2023-06-15 16:17:34,528 - asyncua.client.client - INFO - connect
2023-06-15 16:17:34,529 - asyncua.client.ua_client.UaClient - INFO - opening connection
2023-06-15 16:17:34,646 - asyncua.uaprotocol - INFO - updating client limits to: TransportLimits(max_recv_buffer=8192, max_send_buffer=8192, max_chunk_count=256, max_message_size=2097152)
2023-06-15 16:17:34,647 - asyncua.client.ua_client.UASocketProtocol - INFO - open_secure_channel

AndreasHeine, 2023-06-15:
could you post the debug output?

guess the error is here:
```python
tag_fullname.set_value(78.5)
``
maybe look into this:
https://github.com/AndreasHeine/opcua-tutorial/blob/main/client/write.py