FreeOpcUa/opcua-asyncio #1338: How can I write value to the specific index of the array

TomasRychy, 2023-06-16:
Hello,

I need write a value to the multi-dimensional array to specific index.
I know how to overwrite whole array with this specific value inside of the new value of the array, but I don't know how to write the value to the specific index without overwriting of whole array.
Can you tell me how I can do that?

Thank you,
Tomas

TomasRychy, 2023-06-16:
P.S.: I've tried this, but it returns code BadIndexRangeNoData.
await var.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(new_value, ua.VariantType.Int16, is_array=True)), "1")

TomasRychy, 2023-06-16:
My other question is, if it is possible set values like subrange of array.

schroeder-, 2023-06-16:
On thing is if the opc ua server support index ranges. Not all do. 
Did you try to change a single value via uaexpert? But I am not sure if ua expert is just sending the updatet array.

TomasRychy, 2023-06-18:
I try it on OPC UA server which should be able to change some subrange if the array. My problem is that I don't know how to do it in Python. Can you help me with that?

TomasRychy, 2023-06-19:
UAExpert supports only writing to one specific array index. I need to write values to subrange of array.

schroeder-, 2023-06-19:
I never used it, but this is how it should work:

Write value 10 -> 20

```
new_value = [i+100 for i in range(9)]
await v_ro.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(new_value, ua.VariantType.Int16, is_array=True)), '10:20')
```
write a single value at pos 30
```

new_value = [1]
await v_ro.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(new_value, ua.VariantType.Int16, is_array=True)), '30')
```

schroeder-, 2023-06-19:
This is how the string should be formated from the spec: 
[Spec](https://reference.opcfoundation.org/Core/Part4/v104/docs/7.22)

TomasRychy, 2023-06-19:
Thank you! It looks like it works. Can you please help how can I do it for 2D array? It returns to me error BadIndexRangeNoData. But maybe it is the bug on the side of the server. I'm investigating it.

new_value = [2]
await var.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(new_value, ua.VariantType.Int16, is_array=True)), '0,0')

schroeder-, 2023-06-19:
```
new_value = [2]
await var.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(new_value, ua.VariantType.Int16, is_array=True)), '0')
```

TomasRychy, 2023-06-19:
No it works. My code looks like this:

`new_value = [[0, 0], [0, 0]]
            await var.write_attribute(ua.AttributeIds.Value, ua.DataValue(ua.Variant(new_value, ua.VariantType.Int16, is_array=True)), '1:2,1:2')`

You can close the issue :)

closed 2023-06-19 by schroeder-

TomasRychy, 2023-06-19:
Thank you for your support :)