FreeOpcUa/opcua-asyncio #1353: MAX_OPC_FILETIME limit changed in 1.0.2

robgom, 2023-06-27:
**Describe the bug** <br />
After upgrading asyncua to version 1.0.2, including commit 4e2389e824f951688ff2a0b29b7b994dc409c9b5, tests for OPC UA implementation regarding date time stopped working.
The reason was that condition in win_epoch_to_datetime was changed to lower value (basically replacing leading 9s with 0s).

**Version**<br />
Python-Version: 3.8, 3.9, ...
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.2

Sorry about very short problem description - I'll try to update that myself or ask coworkers to do that.

schroeder-, 2023-06-27:
This is acording the opc ua spec:  [Spec](https://reference.opcfoundation.org/Core/Part6/v104/docs/5.2.2.5)

schroeder-, 2023-06-27:
See #1155