FreeOpcUa/opcua-asyncio #1364: sync.Clinet is not reusable

okapies, 2023-07-18:
**Describe the bug**
Call `connect()` on `sync.Client` after `disconnect()` will raise `ThreadLoopNotRunning` error.

**To Reproduce**<br />
```python
>>> client = Client('opc.tcp://localhost')
>>> client.connect()
>>> client.disconnect()
>>> client.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/topaz/venv/lib/python3.10/site-packages/asyncua/sync.py", line 95, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/user/topaz/venv/lib/python3.10/site-packages/asyncua/sync.py", line 51, in post
    raise ThreadLoopNotRunning(f"could not post {coro}")
asyncua.sync.ThreadLoopNotRunning: could not post <coroutine object Client.connect at 0x7faf0b51ace0>
```

**Expected behavior**
(Re-)initialize `tloop` every time we call `connect()`.

**Version**
- Python-Version: 3.10
- opcua-asyncio Version (e.g. master branch, 0.9): 1.0.2

oroulet, 2023-07-18:
This is really common. You cannot restart a Thread object. But the error message is is really hard to understand...we need to improve that..

okapies, 2023-07-18:
Yes, but the async `Client` is reusable instead. This is problematic especially when want to combine both versions. Can we fix to start the thread when `connect` instead of `__init__`?

miki5799, 2023-11-06:
Same is happening to me and I agree with your proposed expected behaviour.

aclementptw, 2023-11-07:
I'd like to bump this, as we are currently migrating from opcua to opcua-asyncio using the wrapper functions.
We do reuse `Client`s and this poses a problem for us.

miki5799, 2023-11-23:
> Yes, but the async `Client` is reusable instead. This is problematic especially when want to combine both versions. Can we fix to start the thread when `connect` instead of `__init__`?

Hello everyone, 

I have submitted a PR, in which a new thread is started analogous to `__init__` iff, the ThreadLoop has stopped, so that the updated `connect` method looks like this:
```python
def connect(self) -> None:
    if not self.tloop.is_alive():
        self.tloop = ThreadLoop()
        self.tloop.start()
        self.close_tloop = True
    self.tloop.post(self.aio_obj.connect())
```
Analogous for the Server's `start` method.