FreeOpcUa/opcua-asyncio #1377: Data change events are not arriving according to the specified subscription interval

hirsche, 2023-07-27:
**Description** <br />
I am subscribing to node changes of an (non asyncua) OPC UA server. No matter what I am setting (see code below) for `subscription_interval_ms` the datachange_notification handler is roughly called every second, and not as specified in `subscription_interval_ms`. I am connecting to an OPC UA server of a simulation software and when connecting with [UaExpert ](https://www.unified-automation.com/de/produkte/entwicklerwerkzeuge/uaexpert.html)smaller intervals are possible. So it does not seem like that this restriction comes from the server.

Could you kindly support and provide help to check if this is a bug or just wrong usage. If you need more information kindly let me know.

Thanks in advance!

**Code To Reproduce**<br />
```python
# sample value 200ms, but could be any other as well
subscription_interval_ms = 200

# init subscription
subscription = await client.create_subscription(
	subscription_interval_ms, log)

# nodes contains the list of nodes to subscribe
await subscription.subscribe_data_change(nodes)
```

**Expected behavior**<br />
data changes notified in the provided subscription interval

**Change Events Excerpt**<br />
2023-07-27 21:09:10.576800,ns=6;s=::injet_u_pl:MpInjectPlastificationBasic_0.Info.Torque,0.018784865736961365
2023-07-27 21:09:11.578400,ns=6;s=::injet_u_pl:Position_injection,17.79
2023-07-27 21:09:11.578400,ns=6;s=::injet_u_pl:Position_plastification,15276415.65
2023-07-27 21:09:11.578400,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Position,17.799474716186523
2023-07-27 21:09:11.578400,ns=6;s=::injet_u_pl:MpInjectPlastificationBasic_0.Info.Torque,0.018771233037114143
2023-07-27 21:09:12.578400,ns=6;s=::injet_u_pl:Position_injection,18.28
2023-07-27 21:09:12.578400,ns=6;s=::injet_u_pl:Position_plastification,15276425.65
2023-07-27 21:09:12.578400,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Position,18.28558921813965
2023-07-27 21:09:12.578400,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Velocity,0.4861224293708801
2023-07-27 21:09:13.579200,ns=6;s=::injet_u_pl:Position_injection,18.77
2023-07-27 21:09:13.579200,ns=6;s=::injet_u_pl:Position_plastification,15276435.67
2023-07-27 21:09:13.579200,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Position,18.772483825683594
2023-07-27 21:09:13.579200,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Velocity,0.48611313104629517
2023-07-27 21:09:13.579200,ns=6;s=::injet_u_pl:MpInjectPlastificationBasic_0.Info.Torque,0.01878511533141136
2023-07-27 21:09:14.580000,ns=6;s=::injet_u_pl:Position_injection,19.25
2023-07-27 21:09:14.580000,ns=6;s=::injet_u_pl:Position_plastification,15276445.67
2023-07-27 21:09:14.580000,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Position,19.25859832763672
2023-07-27 21:09:15.580800,ns=6;s=::injet_u_pl:Position_injection,19.740000000000002
2023-07-27 21:09:15.580800,ns=6;s=::injet_u_pl:Position_plastification,15276455.68
2023-07-27 21:09:15.580800,ns=6;s=::injet_u_pl:MpInjectBasic_0.Info.Position,19.745492935180664
2023-07-27 21:09:15.580800,ns=6;s=::injet_u_pl:MpInjectPlastificationBasic_0.Info.Torque,0.018774066120386124

**Version**<br />
Python-Version: 3.10.6
opcua-asyncio Version (e.g. master branch, 0.9): asyncua==0.9.95

schroeder-, 2023-07-28:
I would recomend updating the library and try it again. Version 0.9.95 is 1 year old.

Also I will only datachange_notification will only get called when a value changes. Also you need to set the sampling interval for each monitored item to be at least the subscription interval.

hirsche, 2023-07-28:
Thanks @schroeder-, it seems that your hints fixed the issue. I updated the library and changed the sampling interval. Can't say exactly what fixed the issue but it did. Thanks!

closed 2023-07-28 by hirsche