FreeOpcUa/opcua-asyncio #1379: SubHandler.datachange_notification reports incorrect node when multiple subscriptions are opened

giokara-oqton, 2023-07-31:
**Describe the bug** <br />

When creating two subscriptions, `datachange_notification` does not report the correct node as source of the change.

**To Reproduce**<br />

Create a `server.py` with two variables and regularly `write_value` to the nodes, e.g.:

```python
import asyncio
import asyncua
import datetime

async def serve():
    server = asyncua.Server()
    await server.init()
    server.set_endpoint('opc.tcp://0.0.0.0:5840/freeopcua/server/')
    async with server:
        uri = "http://examples.freeopcua.github.io"
        idx = await server.register_namespace(uri)
        myobj = await server.nodes.objects.add_object(idx, "MyObject")
        current_value = 0
        myvar = await myobj.add_variable(idx, "MyVariable", current_value)
        myvar2 = await myobj.add_variable(idx, "MyVariable2", current_value)
        while True:
            print(f'[{datetime.datetime.now().isoformat()}] writing value {2 * current_value} to MyVariable')
            await myvar.write_value(2 * current_value)
            await asyncio.sleep(1)
            print(f'[{datetime.datetime.now().isoformat()}] writing value {2 * current_value + 1} to MyVariable2')
            await myvar.write_value(2 * current_value + 1)
            current_value += 1
            await asyncio.sleep(1)

if __name__ == '__main__':
    asyncio.run(serve())
``` 

Create a `client.py` that subscribes to the two variable nodes and prints some text on change notifications:

```python
import asyncio
import asyncua
import datetime

class SubscriptionHandler(asyncua.common.subscription.SubHandler):
    async def datachange_notification(self, node, value, data):
        print(f'[{datetime.datetime.now().isoformat()}] node {(await node.read_display_name()).Text}: value = {value}')

    def event_notification(self, event):
        print(f'[{datetime.datetime.now().isoformat()}] New event', event)

async def listen():
    async with asyncua.Client(url='opc.tcp://0.0.0.0:5840/freeopcua/server/') as client:
        subscriptions = []
        for child in await (await client.nodes.objects.get_child('2:MyObject')).get_children():
            handler = SubscriptionHandler()
            subscription = await client.create_subscription(10, handler)
            await subscription.subscribe_data_change(child)
            subscriptions.append(subscription)
        while True:
            await asyncio.sleep(0.1)

if __name__ == '__main__':
    asyncio.run(listen())
```

**Expected behavior**<br />

Expect the node in `datachange_notification` to be the node that triggered the datachange notification. That is the case on startup of the client. After that, the same node is always report, irrelevant of the datachange.

**Screenshots**<br />

[image]

**Version**<br />
Python-Version: 3.10.12
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.2

schroeder-, 2023-07-31:
You are only writing to variable 1 so there is no change variable 2....

``` Python
        while True:
            print(f'[{datetime.datetime.now().isoformat()}] writing value {2 * current_value} to MyVariable')
            await myvar.write_value(2 * current_value)
            await asyncio.sleep(1)
            print(f'[{datetime.datetime.now().isoformat()}] writing value {2 * current_value + 1} to MyVariable2')
            await myvar2.write_value(2 * current_value + 1)  # Write to variable2
            current_value += 1
            await asyncio.sleep(1)

```

giokara-oqton, 2023-07-31:
Bad Monday I guess 🤦 Seems to work fine indeed...

This was a minimal test I wrote while having a similar issue in a more involved implementation (where the server is not controlled by me). So will need to further look into the actual implementation.

closed 2023-07-31 by giokara-oqton