FreeOpcUa/opcua-asyncio #1381: load_data_type_definitions crashes due to unexpected OptionSet data from server

loehnertj, 2023-08-01:
**Describe the bug** <br />
I'm using asyncua to connect to a [Tani OPC-UA Gateway](https://www.tanindustrie.de/), version 2.7.3.

When using ``client.load_data_type_definitions``, an ``AttributeError`` is raised from ``structures104.py``.
Investigating the issue, I found that on the Tani Server, there are ``OptionSet`` types in which the ``DataTypeDefinition`` is set as ``StructureDefinition`` instance. Its fields are ``StructureField`` instances. ``structure104.make_enum_code`` raises because the latter are missing the ``Value`` attribute. **Server data + Traceback see below.**

I can't tell yet whether the server or the client is at fault here. I just know that UaExpert connects to the server without any complaints whatsoever. The OPC UA reference docs left me in total confusion on how actual OptionSet subtypes are supposed to look like.  (Any hints are appreciated.)

As far as I see, the data types in question are "builtin" to the Tani Server (something to do with BACnet), so they will always be present and the call will crash consistently. Meaning that ``asyncio`` is currently incompatible with that server, at least regarding structured datatypes.

**To Reproduce**<br />

1. Set up a server with an OptionSet subtype as in the screenshot below. (here: Tani OPC UA Server 2.7.3)
2. Connect with asyncua
3. Call ``client.load_data_type_definitions``.

**Expected behavior**<br />

*If* the server data is within what the standard allows (*opinion wanted*), asyncua should be able to understand it.

Independent from that, it would be good if ``load_data_type_definitions`` would handle such a case more gracefully. For instance, it could log an error message and proceed with the next item instead of crashing altogether.

This would mean introducing some generic try/catch logic in the loops within ``load_data_type_definitions`` and ``load_enums``. One could even collect the exceptions on-the-go and raise them as ``ExceptionGroup`` after completion, if the python version allows. I'd be happy to prepare a PR if it makes sense.

**Screenshot / Traceback**<br />

Here's one representative OptionSet Node on the server: (Path: ``Types / DataTypes / BaseDataType / Structure / OptionSet``):
[image]
Particularly, note that Attribute ``DataTypeDefinition.Fields[0]`` does not have the ``Value`` attribute required by ``make_enum_code``.

Here's the traceback:

    Traceback (most recent call last):
      File "python\lib\site-packages\plc_bridge\asyncua\reconnecting_opcua_client.py", line 98, in aservice
        await self._setup_connection(s)
      File "python\lib\site-packages\plc_bridge\asyncua\reconnecting_opcua_client.py", line 153, in _setup_connection
        self._client = await estack.enter_async_context(
      File "contextlib.py", line 556, in enter_async_context
      File "contextlib.py", line 175, in __aenter__
      File "python\lib\site-packages\plc_bridge\asyncua\extensions.py", line 119, in client_context
        await client.load_data_type_definitions()
      # -- entering library code here -- #
      File "python\lib\site-packages\asyncua\client\client.py", line 833, in load_data_type_definitions
        return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
      File "python\lib\site-packages\asyncua\common\structures104.py", line 461, in load_data_type_definitions
        new_objects.update(await load_enums(server, server.nodes.option_set_type, True))  # also load all optionsets
      File "python\lib\site-packages\asyncua\common\structures104.py", line 549, in load_enums
        env = await _generate_object(name, edef, enum=True, option_set=option_set)
      File "python\lib\site-packages\asyncua\common\structures104.py", line 294, in _generate_object
        code = make_enum_code(name, sdef, option_set)
      File "python\lib\site-packages\asyncua\common\structures104.py", line 532, in make_enum_code
        value = sfield.Value if not option_set else (1 << sfield.Value)
    AttributeError: 'StructureField' object has no attribute 'Value'

**Version**<br />
Python-Version: 3.9.4<br />
opcua-asyncio Version (e.g. master branch, 0.9): tested with 1.0.3, 1.0.1 <br/>
Tani Server 2.7.3

schroeder-, 2023-08-01:
OptionSet is a bit of a beast. Because there are some servers represtining them as a Enum, which is if I rember correct, from a older spec? To support both version we need to check if the value is StructureDefinition and add a new code path to create a option set from StructureDefinition.

loehnertj, 2023-08-01:
It's really not clear from the standard. The only statement I found is in the description of [EnumDefinition](https://reference.opcfoundation.org/Core/Part3/v105/docs/8.50): "This Structured DataType is used to provide the metadata for a custom Enumeration or OptionSet DataType."

Whether to understand this as a "must" regarding the type of `DataTypeDefinition` is a matter of interpretation. Maybe it is good to be somewhat generous here, maybe spiced with a warning message.

When implementing this, the main question is how to assign bit positions in absence of an explicit definition (the "Value"). The obvious thing is to read the fields array and assign to 0x1, 0x2, 0x4... in order.

What do you think about "skipping" invalid definitions and continuing the loop?

schroeder-, 2023-08-01:
I think skipping would be the first step. I did never use OptionsSets, so if someone with a usecase comes around, we can implemented it, so we have a tester to verify the code.

loehnertj, 2023-08-02:
I created two PR's implementing the discussed solutions. See there for some discussion and caveats.

Actually I don't need those OptionSets either, but the error stops me from reading the structures I need ;-)

When writing the tests, I noted that the tests in ``test_common.py`` are not isolated for each other. Namely, `load_enums` creates new classes e.g. `ua.MyEnum` *lazily*. If any Enum is already there from a previous test, it is not overwritten, which can cause unexpected failure or worse, unexpected *pass* of a test, depending on whether a particular previous test was run or not. I didn't pursue this further, but thought you might want to have a look.

closed 2023-09-29 by schroeder-