FreeOpcUa/opcua-asyncio #1387: Call Method Arguments unclear

oskarmue, 2023-08-02:
I use opcua-asyncio to connect to a opc ua server hosted by an IO-Link-master.
With this opc ua server I wish to change the ISDU of a sensor connected to the IO-Link-master.
In order to do that I try to call a method (with two arguments - index and subindex - the index is supposed to be an Uint16 and the subindex is supposed to be in byte) provided on the server.
When I do this "manually" while connected to the server with a Prosys client, the method works fine.

Unfortunately I am not able to call the method with this library. 
I always receive an error that the one of the arguments is not an integer.
This is weird, because only the sonc argument should be an byte (as mentioned above).
And even more weird, if I define the second argument as an integer as well, I get an error, that one or more arguments are missmatched.
Appended is the code I use and a picture of the method call on prosys.
Is there something I am doing wrong or is this kind of method call not implemented yet?

Code:
```
    parent_object = client.get_node("ns=3;i=5002")
    method = client.get_node("ns=3;i=7005")
    arg1 = ua.Variant(40, ua.VariantType.UInt16)
    arg2 = ua.Variant('00000000', ua.VariantType.Byte)
    output = parent_object.call_method(method, [arg1, arg2])
```

![Screenshot (3)](https://github.com/FreeOpcUa/opcua-asyn
[image]
cio/assets/73947915/1483b61c-345c-4820-8ced-59d114bdd40a)

oroulet, 2023-08-03:
And the error is? Show the stack trace 
Also you can use Wireshark to see exactly what is sent from both clients

schroeder-, 2023-08-03:
Byte is a 8bit Integer, not a String, so i would create the args like this:

```
 arg1 = ua.Variant(ua.Int16(40))
 arg2 = ua.Variant(ua.Byte(0))
```

oskarmue, 2023-08-03:
the stack traces look as follows:

when I try 
`arg1 = ua.Variant(40, ua.VariantType.Uint16)
arg2 = ua.Variant('00000000', ua.VariantType.Byte) #-> same error as when arg2 = ua.Variant(b'00000000', ua.VariantType.Byte)
`
[image]

When I try, as suggested by schroeder-

`arg1 = ua.Variant(ua.VariantType.Uint16(40))
arg2 = ua.Variant(ua.VariantType.Byte(0))
`
[image]

But most interesting, when trying

`arg1 = ua.Variant(40, ua.VariantType.Uint16)
arg2 = ua.Variant(00000000, ua.VariantType.Byte)
`
[image]

schroeder-, 2023-08-03:
You made a typo it is not ua.VariantType.Byte!
```Python 
arg1 = ua.Variant(ua.Uint16(40)) 
arg2 = ua.Variant(ua.Byte(0)) 
```
Also i think you should not pass a list, pass them as seperate args:
```Python
arg1 = ua.Variant(ua.Uint16(40)) 
arg2 = ua.Variant(ua.Byte(0)) 
output = parent_object.call_method(method, arg1, arg2)
```

oskarmue, 2023-08-03:
Oh ja right. I tried it with this
'
arg1 = ua.Variant(ua.Uint16(40)) 
arg2 = ua.Variant(ua.Byte(0)) 
output = parent_object.call_method(method, arg1, arg2)
'
now I get the following error message:

[image]

schroeder-, 2023-08-03:
Calling now works. This is the respone from the server.
Looks like a internal error from the server, maybe the arguments are invalid?

closed 2023-08-18 by oskarmue