FreeOpcUa/opcua-asyncio #1388: Problem reading arrays inside an ExtensionObject variable

ThomasDegallaix, 2023-08-03:
First of all, thank you very much for your hard work !

We are trying to send data from an OPCUA server embedded in a PLC using Codesys to an embedded computer acting as a client. Sending data generally works fine, except when we decide to send more complex data containing arrays, which take the form of structures in Codesys and are translated into ExtensionObjects when using the OPCUA protocol.

When we check the contents of the variables using the UAExpert client, the arrays are complete and can be read without any problems. However, when these same variables are retrieved by the OPCUA asyncio client on the embedded computer, the arrays are no longer treated as such and data is lost. We retrieve the data by subscribing to the nodes and storing them in a queue once they have been received in the callback notifying us of data modifications.

A test was carried out with 2 ExtensionObjects with arrays of 10 Int16 containing different data. 
[image]
[image]

Here are the results when reading the variables using the UAexpert client :

[image]

[image]

And here are the results when reading the variables using the library :

[image]

[image]

After displaying the contents, it can be seen that the size of the 1st array is correctly read, but the contents are limited to the value of the data at the beginning of the list. Then, the size of the next list corresponds to the addition of the 4 bytes of the next 2 indexes (concatenating the binary of 2 and 3 gives 196610 in Int32 decimal writing, and the same applies to the second list with 12 and 13, which gives 851980 in Int32). Finally, the contents of the 2nd list correspond to the contents of the 4th value of the 1st list. The data is therefore decoded as a succession of bytes, irrespective of the fact that they are lists.

Interestingly, if I call the read_value() function directly on the ExtensionObject field containing the array, the data is correctly decoded.

We are currently running python 3.8.10 and opcua asyncio 1.0.3

Thanks in advance !

schroeder-, 2023-08-03:
Did you call load_type_definitions or load_data_type_definitions to load the datatypes?

ThomasDegallaix, 2023-08-03:
Yes, I tried both. Only load_type_definitions() allows me to decode correctly most of the data except arrays embedded inside an ExtensionObject.

schroeder-, 2023-08-03:
Then the server doesn't support the new DataType definitions. 
load_type_definitions is not complete, maybe you can extract the bsd of the datatype? 
This is done if go under Root -> Types -> DataTypes -> OPC Binary , there you can extract the value of the top level nodes (Dictonarys), as binary string?

ThomasDegallaix, 2023-08-03:
Thank you for the advice !
When I reach OPC Binary, I can see all our ExtensionObjects. 
[image]

However when I select the datatype used during our test and check the attributes and specifically the field value, it only gives me datatype.
[image]

Is there any other way to get the bsd ?

schroeder-, 2023-08-03:
The bsd is in the top level variable "CODESYSSPV3/S3/LecVarAccess"

ThomasDegallaix, 2023-08-03:
After converting the hex value of the bsd to a readable format I got this :
```
<opc:StructuredType BaseType='ua:ExtensionObject' Name='ecomat_display.StructTest2'>
<opc:Field TypeName='opc:Int16' Name='var1'/>
<opc:Field TypeName='opc:Int16' Name='var2'/>
<opc:Field TypeName='opc:Int32' Name='__test_list1Length'/>
<opc:Field TypeName='opc:Int16' LengthField='__test_list1Length' Name='test_list1'/>
<opc:Field TypeName='opc:Int32' Name='__test_list2Length'/>
<opc:Field TypeName='opc:Int16' LengthField='__test_list2Length' Name='test_list2'/>

</opc:StructuredType><opc:StructuredType BaseType='ua:ExtensionObject' Name='ecomat_display.StructTest1'>
<opc:Field TypeName='opc:Int16' Name='var1'/>
<opc:Field TypeName='opc:Int16' Name='var2'/>
<opc:Field TypeName='opc:Int32' Name='__test_list1Length'/>
<opc:Field TypeName='opc:Int16' LengthField='__test_list1Length' Name='test_list1'/>
<opc:Field TypeName='opc:Int32' Name='__test_list2Length'/>
<opc:Field TypeName='opc:Int16' LengthField='__test_list2Length' Name='test_list2'/>
```
I'm not sure about what should be printed in case of fields containing arrays but it seems to follow the original format.

schroeder-, 2023-08-03:
Ok our passer doesn't support that syntax currently. A Length Field has to have to start with "NoOf", instead of end with "Length".

ThomasDegallaix, 2023-08-03:
Thank you very much for your explanation !
Do you think there is anything I could do to make it work ? However, I don't think there is anything I can do on the Codesys side.

schroeder-, 2023-08-03:
I don't think you can change anything on the codesys side. But I can make a pr, soon. It is only a smal change.

ThomasDegallaix, 2023-08-03:
Thank you for your involvement,  I can't wait to see the modification. It will be really useful for us !

schroeder-, 2023-08-06:
Made a pr #1395 let me know if it works for you.

ThomasDegallaix, 2023-08-07:
It works like a charm, thank you !

PR #1395 (2023-08-09): Support pre 1.04 struct length fields for codesys
Fixes #1388
closed 2023-08-09 by oroulet