FreeOpcUa/opcua-asyncio #1405: read_values random timeout

gadamatti, 2023-08-16:
When using the read_values method and there is a bad quality tag in a list, from an existing channel/device on an OPC Ua server, sometimes the timeout occurs, sometimes it skips altogether.

Opposed to that, if I try to read a list containing a tag from a non-existent channel/device, the tag gets completely ignored on when trying to read its value.

When the timeout is triggered, it seems to keep on happening for some random amount of time, and then it suddenly stops happening altogether.

The exception's class that happens is: asyncio.exceptions.TimeoutError.

I'm also using a Client timeout of 2s.

**To Reproduce**<br />
- Configure an OPC Ua server with 2 different devices. One existing and one not existing (both can be simulated, doesn't really matter)
- Use the read_values method to read a list containing valid tags and non-existing tags from the second device.

**Expected behavior**<br />
The timeout shouldn't be randomly occurring. It should always happen, or never happen.

**Version**<br />
Python-Version: 3.10.9
python-opcua Version: 1.0.1
Top Server (OPC Ua server used): 6.12

schroeder-, 2023-08-16:
Do have an example code? This sounds more of an issue mixing threading and asyncio. Or the server is busy and can't handle the request. Both issues that we can not fix in the library.

gadamatti, 2023-08-16:
This is the implementation that we did using read_values() while adding some management when that exception pops-up. So that we end up nullifying only the list that contains the problematic address, while maintaining the rest of the code running.

```
async def read_many(self, addresses:list):

        none_tags_counter = 0
        none_tags_list = []
        nodes = []
        read_values = []

        try:
            async with Client(self.host) as self.client:

                # Get all node objects
                for address in addresses:

                    nodes.append(self.client.get_node(self.check_address(address)))

                read_values = await self.client.read_values(nodes)

                # Checks for wrong addresses
                for i in range(len(addresses)):
                    if read_values[i] == None:
                        none_tags_counter += 1
                        none_tags_list.append((addresses)[i])

                # Logs wrong addresses found
                if none_tags_counter != 0:
                    app_logger.warning(f"{none_tags_counter} addresses has None value. These addresses may not be right: {none_tags_list}.")

                return read_values

        except Exception as exc:

            if exc.__class__ == asyncio.exceptions.TimeoutError:
                app_logger.info(f"Timeout while trying to read following list of tags: {addresses}")
            else:
                raise Exception("OPC Ua server generic exception (read_many method)")

            return [None]*len(addresses)
```

schroeder-, 2023-08-18:
Possible depends on the server. Did you try to split the read in 50 values per call? Because the 2 second timeout is for the total call. Means 2 seconds to get all 150+ values.

closed 2023-08-18 by schroeder-

gadamatti, 2023-08-18:
Actually those +-150 tags are already split into 5 different lists. 4 of them are small, and one of them has 100+ tags, and the problem was confirmed on both the big list and on the small ones.

Again, this only happens in the specific scenario when we are trying to read invalid tags from an existing channel/device.

Shouldn't the library be consistent on the behavior of timeout always/never happening?

schroeder-, 2023-08-18:
The library is using the same timeout every where. So this is either asyncio  or the targeted server.

schroeder-, 2023-08-18:
Maybe do a wireshark trace, so you can see what is going on, on the wire.