FreeOpcUa/opcua-asyncio #1406: Server does not support requests specifying ExpandedNodeId

okapies, 2023-08-21:
**Describe the bug** <br />
`Server` accepts `ExpandedNodeId` as a parameter in a request, but it doesn't match with the node added to the server when its nodeid is `NodeId` (i.e. registered by using namespace index instead of namespace URI).

**To Reproduce**<br />
```python
import asyncio
from asyncua import ua, Server

async def run_server():
    server = Server()
    await server.init()

    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")
    server.set_server_name("FreeOpcUa Example Server")
    server.set_security_policy([ua.SecurityPolicyType.NoSecurity])

    idx = await server.register_namespace("http://examples.freeopcua.github.io")
    await server.nodes.objects.add_variable(
        ua.NodeId("MyVariable", idx), ua.QualifiedName("MyVariable", idx), 6.7
    )
    async with server:
        while True:
            await asyncio.sleep(0.1)
 
asyncio.run(run_server())
```

```python
from asyncua import ua
from asyncua.sync import Client

>>> c = Client('opc.tcp://localhost:4840')
>>> c.connect()
>>> c.nodes.namespace_array.read_value()
['http://opcfoundation.org/UA/', 'urn:freeopcua:python:server', 'http://examples.freeopcua.github.io']
>>> c.get_node("ns=2;s=MyVariable").read_value()
6.7
>>> c.get_node("nsu=http://examples.freeopcua.github.io;s=MyVariable").read_value()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  ...
  File "/home/user/project/venv/lib/python3.10/site-packages/asyncua/ua/uatypes.py", line 372, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadNodeIdUnknown: "The node id refers to a node that does not exist in the server address space."(BadNodeIdUnknown)

```

**Expected behavior**<br />
Normalize the received requests by converting `nsu` into `ns`.

**Screenshots**<br />
n/a

**Version**<br />
- Python-Version: 3.10
- opcua-asyncio Version: 1.0.3

AndreasHeine, 2023-08-21:
according to the opc ua reference: 
for the requests you need to use the NodeId-Format only!

the ExpandedNodeId-Format is for storing configurations/nodeid's!
after client connecting you need to read the namespace array and resolve the nsu (ns-uri) to ns (ns-idx). the reason for that is that the ns index is not an constant and might change over time as models get added or changed... the only constant is the nsu (the actual unique name in the server) to identify the namespace and within the namespace the constant is the identifier (which is unique within the namespace)

AndreasHeine, 2023-08-21:
e.g. :

okapies, 2023-08-22:
Yes, ns-idx is not realiable to access the nodes. Should we support ns-uri in the request like [QuickOpc](https://opclabs.doc-that.com/files/onlinedocs/QuickOpc/Latest/User's%20Guide%20and%20Reference-QuickOPC/OPC%20UA%20Node%20IDs.html)? It seems we need to normalize the given `NodeId` before reading/writing a node to the address space:

https://github.com/FreeOpcUa/opcua-asyncio/blob/fa73b116cd9ee2e9d00c9caf2e5f8784d36cdb82/asyncua/server/address_space.py#L79-L80

https://github.com/FreeOpcUa/opcua-asyncio/blob/fa73b116cd9ee2e9d00c9caf2e5f8784d36cdb82/asyncua/server/address_space.py#L107-L109

Otherwise, `ExpandedNodeId` should always be rejected as invalid in the `Server`.