FreeOpcUa/opcua-asyncio #1417: I want to make asyncua a Windows service, but it started with an error

tianyao1111, 2023-08-28:
**Describe the bug** <br />
I want to make asyncua a Windows service, but it started with an error

**To Reproduce**<br />
server-methods2.py:
```
#!/usr/bin/env python
# coding: utf-8
import os
import sys
import subprocess
import configparser
import asyncio
import logging
#导入pandas包
import pandas as pd
from asyncua import ua, uamethod, Server
config = configparser.ConfigParser() #使用工具类读写ini配置文件
current_path=os.path.dirname(os.path.realpath(__file__))
config.read(os.path.join(current_path, r'server.ini'),encoding="gbk")

#server的启动端口
port = config['server']['port']
#灰度格框条数
gray_grid_boxes_num=int(config['data-analysis']['gray_grid_boxes_num'])
#拍照间隔时间
per_col_time=int(config['data-analysis']['per_col_time'])
#每格高度
per_line_height=int(config['data-analysis']['per_line_height'])
#找截止行需去掉的末尾行数
endline_num=int(config['data-analysis']['endline_num'])
#找截止行需去掉的末尾行数
startline_num=int(config['data-analysis']['startline_num'])

def calculate_func(excel_name):
    #读取表格数据
    excel_path=excel_name
    origin_data=pd.read_excel(excel_path,header=None,sheet_name=0)
    #do something 
    return ratio,slope

class SubHandler(object):
    async def datachange_notification(self, node, val, data):
        if val:
            try:
                ratio,slope=calculate_func(val)
            except Exception as e:
                ratio,slope=-1.0,-1.0
                print("计算报错："+str(e))
            idx=node.nodeid.NamespaceIndex
            parent = await node.get_parent()
            ratioNode=await parent.get_child(str(idx)+":ratio") 
            await ratioNode.write_value(ratio) 
            slopeNode=await parent.get_child(str(idx)+":slope")
            await slopeNode.write_value(slope) 
            
async def main():
    logging.basicConfig(level=logging.ERROR)

    # now set up our server
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:"+port+"/freeopcua/server/")
    server.set_server_name("FreeOpcUa Example Server")

    # set up our own namespace
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    myobj = await server.nodes.objects.add_object(idx, "MyObject")
    excel_name = await myobj.add_variable(idx, "excel_name", "")
    await excel_name.set_writable()
    ratio = await myobj.add_variable(idx, "ratio", 0.0)
    slope = await myobj.add_variable(idx, "slope", 0.0)

    handler = SubHandler()
    sub = await server.create_subscription(100, handler)
    await sub.subscribe_data_change(excel_name)
    

    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    with subprocess.Popen([os.path.join(current_path, r"killPort.bat"),port], stdout=subprocess.PIPE) as  process:
        process.wait()
    for i in range(30):
        cmd = [os.path.join(current_path, r"checkKillPort.bat"),port]
        with subprocess.Popen(cmd, stdout=subprocess.PIPE) as  process:
            process.wait()
            out=process.stdout.read()
            process.stdout.close()
            if out.strip().decode()=="0":
                break
            else:
                time.sleep(2)
    asyncio.run(main())

```

makeService2.py:
```
import os
import sys
import servicemanager
import win32event
import win32service
import win32timezone
import win32serviceutil
# import subprocess

class MyService(win32serviceutil.ServiceFramework):
    _svc_name_ = "TianYaoService"
    _svc_display_name_ = "Tian Yao Service"

    def __init__(self, args):
        win32serviceutil.ServiceFramework.__init__(self, args)
        self.hWaitStop = win32event.CreateEvent(None, 0, 0, None)

    def SvcStop(self):
        self.ReportServiceStatus(win32service.SERVICE_STOP_PENDING)
        win32event.SetEvent(self.hWaitStop)

    def SvcDoRun(self):
        # while True:
        #     time.sleep(2)
        self.main()

    def main(self):
        # # 在这里添加你的 Python 脚本的执行逻辑
        # #杀死端口
        # with subprocess.Popen([PYTHON_EXECUTABLE,killPortPy,killPort,checkPort,port], stdout=subprocess.PIPE) as  process:
        #     process.wait()
       
        # os.system(f"{PYTHON_EXECUTABLE} {SCRIPT_PATH} {port}")
        try:
            os.system(r'python C:\Users\ty\Desktop\extra项目\拟合曲线项目\OPCUA接口\server-methods2.py')
        except Exception as e:
            print(str(e))

if __name__ == '__main__':
    if len(sys.argv) == 1:
        servicemanager.Initialize()
        servicemanager.PrepareToHostSingle(MyService)
        servicemanager.StartServiceCtrlDispatcher()
    else:
        win32serviceutil.HandleCommandLine(MyService)
```
run:
`python makeService2.py install`
then
The Windows service reported an error when starting the service

**Expected behavior**<br />
I want to register the server made of asyncua as a Windows service, with Windows management for server startup and shutdown restart

**Screenshots**<br />
[image]

**Version**<br />
Python-Version:3.10.9<br />
opcua-asyncio Version (e.g. master branch, 0.9):1.0.2

schroeder-, 2023-08-29:
With the provided infos it's not possible to help. Why you think asyncua is the problem? 
The hole code looks strange. Why use os.system to start our code? Why not just call the code? Are there any logs?

tianyao1111, 2023-09-18:
> With the provided infos it's not possible to help. Why you think asyncua is the problem? The hole code looks strange. Why use os.system to start our code? Why not just call the code? Are there any logs?

I changed to directly start the code, but still reported an error. 
code:
```
import os
import sys
import subprocess
import configparser
import asyncio
import logging
#导入pandas包
import pandas as pd
from asyncua import ua, uamethod, Server
config = configparser.ConfigParser() #使用工具类读写ini配置文件
current_path=os.path.dirname(os.path.realpath(__file__))
config.read(os.path.join(current_path, r'server.ini'),encoding="gbk")

#server的启动端口
port = config['server']['port']
#灰度格框条数
gray_grid_boxes_num=int(config['data-analysis']['gray_grid_boxes_num'])
#拍照间隔时间
per_col_time=float(config['data-analysis']['per_col_time'])
#每格高度
per_line_height=float(config['data-analysis']['per_line_height'])
# #找截止行需去掉的末尾行数
# endline_num=int(config['data-analysis']['endline_num'])
#找截止行需去掉的末尾行数
startline_num=int(config['data-analysis']['startline_num'])

def calculate_func(excel_name):
    #do something
    return ratio,slope

class SubHandler(object):
    async def datachange_notification(self, node, val, data):
        if val:
            try:
                ratio,slope=calculate_func(val)
            except Exception as e:
                ratio,slope=-1.0,-1.0
                print("计算报错："+str(e))
            idx=node.nodeid.NamespaceIndex
            parent = await node.get_parent()
            ratioNode=await parent.get_child(str(idx)+":ratio") 
            await ratioNode.write_value(ratio) 
            slopeNode=await parent.get_child(str(idx)+":slope")
            await slopeNode.write_value(slope) 
            
async def main1():
    logging.basicConfig(level=logging.ERROR)

    # now set up our server
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:"+port+"/freeopcua/server/")
    server.set_server_name("FreeOpcUa Example Server")

    # set up our own namespace
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    myobj = await server.nodes.objects.add_object(idx, "MyObject")
    excel_name = await myobj.add_variable(idx, "excel_name", "")
    await excel_name.set_writable()
    ratio = await myobj.add_variable(idx, "ratio", 0.0)
    slope = await myobj.add_variable(idx, "slope", 0.0)

    handler = SubHandler()
    sub = await server.create_subscription(100, handler)
    await sub.subscribe_data_change(excel_name)
    

    async with server:
        while True:
            await asyncio.sleep(1)

import os
import sys
import servicemanager
import win32event
import win32service
import win32timezone
import win32serviceutil
# import subprocess

class MyService(win32serviceutil.ServiceFramework):
    _svc_name_ = "TianYaoService"
    _svc_display_name_ = "Tian Yao Service"

    def __init__(self, args):
        win32serviceutil.ServiceFramework.__init__(self, args)
        self.hWaitStop = win32event.CreateEvent(None, 0, 0, None)

    def SvcStop(self):
        self.ReportServiceStatus(win32service.SERVICE_STOP_PENDING)
        win32event.SetEvent(self.hWaitStop)

    def SvcDoRun(self):
        # while True:
        #     time.sleep(2)
        try:
            self.main()
        except Exception as e:
            print(e)
            raise

    def main(self):
        # 在这里添加你的 Python 脚本的执行逻辑
        with subprocess.Popen([os.path.join(current_path, r"killPort.bat"),port], stdout=subprocess.PIPE) as  process:
            process.wait()
        for i in range(30):
            cmd = [os.path.join(current_path, r"checkKillPort.bat"),port]
            with subprocess.Popen(cmd, stdout=subprocess.PIPE) as  process:
                process.wait()
                out=process.stdout.read()
                process.stdout.close()
                if out.strip().decode()=="0":
                    break
                else:
                    time.sleep(2)
        asyncio.run(main1())
        

if __name__ == '__main__':
    if len(sys.argv) == 1:
        servicemanager.Initialize()
        servicemanager.PrepareToHostSingle(MyService)
        servicemanager.StartServiceCtrlDispatcher()
    else:
        win32serviceutil.HandleCommandLine(MyService)
```
This is an error on the event viewe
```
The instance's SvcRun() method failed 
Traceback (most recent call last):
  File "C:\ProgramData\anaconda3\Lib\site-packages\win32\lib\win32serviceutil.py", line 1062, in SvcRun
    self.SvcDoRun()
  File "C:\Users\ty\Desktop\extra项目\拟合曲线项目\OPCUA接口\2029-9-13修改\makeService2.py", line 143, in SvcDoRun
    self.main()
  File "C:\Users\ty\Desktop\extra项目\拟合曲线项目\OPCUA接口\2029-9-13修改\makeService2.py", line 162, in main
    asyncio.run(main1())
  File "C:\ProgramData\anaconda3\lib\asyncio\runners.py", line 39, in run
    loop = events.new_event_loop()
  File "C:\ProgramData\anaconda3\lib\asyncio\events.py", line 798, in new_event_loop
    return get_event_loop_policy().new_event_loop()
  File "C:\ProgramData\anaconda3\lib\asyncio\events.py", line 688, in new_event_loop
    return self._loop_factory()
  File "C:\ProgramData\anaconda3\lib\asyncio\windows_events.py", line 315, in __init__
    super().__init__(proactor)
  File "C:\ProgramData\anaconda3\lib\asyncio\proactor_events.py", line 639, in __init__
    signal.set_wakeup_fd(self._csock.fileno())
ValueError: set_wakeup_fd only works in main thread of the main interpreter
```

closed 2023-09-18 by schroeder-