FreeOpcUa/opcua-asyncio #1418: check server alive from the server code side

ikasakmongkon, 2023-08-30:
Hi,
I am sorry. I am a newbie 
Is it possible that we can check if the server still alive or crashed or network interface down from the server code side?

Thanks

schroeder-, 2023-08-30:
call `check_connection` which will throw an exception on connection lost.

``` Python
while True:
     await asyncio.sleep(1)
     await client.check_connection()  # Throws a exception if connection is lost
```

ikasakmongkon, 2023-08-30:
Hi, 
Is there an attribute of server status or internal the server status that we can check the server status is running or failed

schroeder-, 2023-08-30:
No this currently not possible.

AndreasHeine, 2023-08-30:
what do you mean with running? tcp socket? processing ua requests? 

this is not so easy to do opc ua has a layered architecture...