FreeOpcUa/opcua-asyncio #1424: Serializer fails when dataclass field has `None`

okapies, 2023-09-05:
**Describe the bug** <br />
`struct_to_binary` raises `TypeError` when a dataclass field has `None` value.

**To Reproduce**<br />
For example, `TargetName` is a `QualifiedName` which is [nullable](https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.2) but asyncua fails to serialize it.

```python
>>> from asyncua import ua
>>> rpath = ua.RelativePath(
  Elements=[
    ua.RelativePathElement(
      ReferenceTypeId=ua.NodeId(ua.ObjectIds.HierarchicalReferences),
      IsInverse=False,
      IncludeSubtypes=True,
      TargetName=None,  # <---
    )
  ]
)
>>> ua.ua_binary.struct_to_binary(rpath)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 336, in struct_to_binary
    return serializer(obj)
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 327, in <genexpr>
    else serializer(obj.__dict__[name])
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 401, in serialize
    return data_size + b''.join(type_serializer(el) for el in val)
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 401, in <genexpr>
    return data_size + b''.join(type_serializer(el) for el in val)
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 327, in <genexpr>
    else serializer(obj.__dict__[name])
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 335, in struct_to_binary
    serializer = create_dataclass_serializer(obj.__class__)
  File "/home/user/opcua-asyncio/asyncua/ua/ua_binary.py", line 284, in create_dataclass_serializer
    data_fields = fields(dataclazz)
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/dataclasses.py", line 1198, in fields
    raise TypeError('must be called with a dataclass type or instance') from None
TypeError: must be called with a dataclass type or instance
```

**Expected behavior**<br />
Encode Python `None` value to `null` in OPC message properly.

**Screenshots**<br />
n/a

**Version**<br />
- Python-Version: 3.10.12
- opcua-asyncio Version (e.g. master branch, 0.9): 1.0.4

schroeder-, 2023-09-05:
It is not nullable see:
https://github.com/FreeOpcUa/opcua-asyncio/blob/4c987a69d52e16e49c2c86d16119ec0cdcb82117/asyncua/ua/uatypes.py#L701-L721
and https://reference.opcfoundation.org/v105/Core/docs/Part4/7.31

okapies, 2023-09-05:
Thanks for the reference. Does *"empty QualifiedName"* mean a `QualifiedName` with `Name=None` or `""`?

schroeder-, 2023-09-05:
Just call `QualifiedName()`.
Also Nullables are just a wire thing. The decode is free to make an nullstring to a empty string. So in 98% of the usecases it doens't matter if a value is Null or Empty.