FreeOpcUa/opcua-asyncio #1425: Deserializer does not fill optional field with `None`

okapies, 2023-09-05:
**Describe the bug** <br />
Deserializer for dataclass does not fill the optional field if it is `None`.

**To Reproduce**<br />
```python
from dataclasses import dataclass, field
from typing import Optional

from asyncua import ua
from asyncua.ua.ua_binary import struct_from_binary, struct_to_binary

@dataclass
class MyStruct:
    Encoding: ua.Byte = field(default=0, repr=False, init=False)
    a: ua.Int32 = 1
    b: Optional[ua.Int32] = 2
    d: ua.Int32 = 4

def main():
    m1 = MyStruct(1, None, 3)
    print(m1)
    data1 = struct_to_binary(m1)
    print(data1)
    m2 = struct_from_binary(MyStruct, ua.utils.Buffer(data1))
    print(m2)

    assert m1 == m2

if __name__ == "__main__":
    main()
```

```
$ python test.py 
MyStruct(a=1, b=None, d=3)
b'\x00\x01\x00\x00\x00\x03\x00\x00\x00'
MyStruct(a=1, b=2, d=3)
Traceback (most recent call last):
  File "/home/usert/test/test.py", line 28, in <module>
    main()
  File "/home/user/test/test.py", line 24, in main
    assert m1 == m2
AssertionError
```

**Expected behavior**<br />
Filled with `None`.

**Screenshots**<br />
n/a

**Version**<br />
- Python-Version: 3.10.12<br />
- opcua-asyncio Version (e.g. master branch, 0.9): 1.0.4

schroeder-, 2023-09-05:
This is not a bug. It always uses the default value. Change your code to this and it will work.
```
@dataclass
class MyStruct:
    Encoding: ua.Byte = field(default=0, repr=False, init=False)
    a: ua.Int32 = 1
    b: Optional[ua.Int32] = None
    d: ua.Int32 = 4
```

okapies, 2023-09-05:
Thanks. I found the description of the behavior.

https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.2

> The Nullable column indicates whether a ‘null’ value exists for the [DataType](https://reference.opcfoundation.org/search/206?t=DataType) in all [DataEncodings](https://reference.opcfoundation.org/search/206?t=DataEncodings). A ‘null’ value is a value that is equavalent ‘no value specified’. A nullable type with a default value means the default value shall be interpreted equivalent to a null.

closed 2023-09-05 by okapies