FreeOpcUa/opcua-asyncio #1430: Client side check_certificate is producing false positives

rth, 2023-09-07:
The current  [`check_certificate`](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/crypto/uacrypto.py#L281) might be a bit too strict on the client side.
 For instance when connecting to an Ignition OPCUA server, I'm hitting 2 failed checks:

 1. certificate does not contain the application uri ({application_uri}). Most applications will reject a connection without it.
 2. certificate does not contain the hostname in DNSNames {hostname}. Some applications will check this.
 
 these will be logged for each new connection as an error, and yet the connection works fine.  There is also no way of disabling these short of monkeypatching.
 
I'm not very knowledgeable about certificate validation best practices, and could be mistaken, but I feel it's not really client's role to do such strong enforcement of certificates. In particular, logging as an error that "Most (or some) applications " will fail with this certificate, does seem to be a false positive when the application works fine.

**Version**<br />
Python-Version: 3.11
opcua-asyncio Version: master

AndreasHeine, 2023-09-07:
opc ua reference has some strict rules on that to avoid security issues e.g. check if the client which is connected has the same uri that is in the subject altname extension of the cert used

schroeder-, 2023-09-08:
Also this is logged because we had lot of problems users, that could not connect to servers.  Also like the message sugguest, nearly all servers will need these parameters. 

Maybe we could log a warning instead of an error.

AndreasHeine, 2023-09-09:
> I understand. +1 to make it a warning (debug or info) so that servers that don't fully respect that part of the spec can still be connected without flooding the logs with these messages.

should be optional to ignore cert issues

oroulet, 2023-09-10:
yes if it does not break things it should be a warning.

bitkeeper, 2023-09-10:
With PR #1415 there is a certificate validator available that can be configured what to check from a certificate.
When the validator is used, maybe should also be used on the locations where now `check_certificate` is used?