FreeOpcUa/opcua-asyncio #1441: load_data_type_definitions crashes on S7-1500 UDT

bauermarkus, 2023-09-14:
I tried to use your example but it crashes at my S7-1500 UDT

```
    client = Client(url=config['opc']['server'])
    client.secure_channel_timeout = config['opc']['timeout']
    client.session_timeout = config['opc']['timeout']

    await client.connect()
    root = client.get_root_node()

    print("Objects node is: ", root)

    # Node objects have methods to read and write node attributes as well as browse or populate address space
    print(f"Children of root are:")
    for c in await root.get_children():
        print(f'\t{c} {await c.read_display_name()} {await c.read_browse_name()}')

    namespaces = await client.get_namespace_array()
    for ns in namespaces:
        print( f'Index of {ns}: {await client.get_namespace_index(ns)}' )

    print("Register namespace")
    idx = await client.register_namespace(namespaces[3])
    print("idx", idx)

    path = f'ns=3;s="S360 Main Data DB"."Offline Nest"[0]."PDR"."LPC"'
    print(f'Create node for {path}')
    node = client.get_node(path)
    dt_id = await node.read_data_type()

    dt = client.get_node(dt_id)
    dtd = await dt.read_data_type_definition()

    value = await node.read_value()
    print("*"*50)
    print(f"Node:{node}")
    print(f'Value:{value}')
    print(f'Datatype:{dt}')
    print(f'Datatypedefinition:{dtd}')
    print()
    print()

    print("*"*50)
    print("Load datatype defintions")
    await client.load_data_type_definitions()
```

[opc.log](https://github.com/FreeOpcUa/opcua-asyncio/files/12611113/opc.log)

That is what Brosys is showing:

[image]
[image]
[image]

I am not quiet good with OPC-UA yet, so I don't know really what I could do to help you to recreate it but let me know and I will try to give you all data you need asap

Also: Can I only create definitions for a subset?

I checked #234 and tried this:

```
    node = client.get_node(path)
    dt_id = await node.read_data_type()

    dt = client.get_node(dt_id)
    dtd = await dt.read_data_type_definition()

    value = await node.read_value() 
```
But I did not understand how to access the data in a convenient pythonic way

schroeder-, 2023-09-15:
You have to call `client.load_data_type_definitions` once, before the read. This loads all Datatypes of the server. After that you have python classes.

```
await client.load_data_type_definitions()
value = await node.read_value() 
print(f"node value: {value}")
```

bauermarkus, 2023-09-15:
> You have to call `client.load_data_type_definitions` once, before the read. This loads all Datatypes of the server. After that you have python classes.
> 
> ```
> await client.load_data_type_definitions()
> value = await node.read_value() 
> print(f"node value: {value}")
> ```

Hello Schröder,

I call `await client.load_data_type_definitions()` but it crashes or you want to tell me, it must be the first thing to do?

AndreasHeine, 2023-09-15:
depending of the settings in the s7 hw-konfig

you should try/except "await client.load_data_type_definitions()"
if it fails its not the OPC UA 1.04 Structures, if an exception happens you need to "await client.load_type_definitions()" which should load the OPC UA 1.03 Structures

schroeder-, 2023-09-15:
Do have a stacktrace or anything because in your log is no error logged.

bauermarkus, 2023-09-15:
> Do you have a stacktrace or anything because in your log is no error logged.

Sorry, I was expecting opc-ua will log it, here is the output

```
Exception has occurred: TypeError       (note: full exception trace is shown but execution is paused at: <module>)
'Field' object is not subscriptable
  File "<string>", line 12, in ST_MediaViewer
  File "<string>", line 4, in <module>
  File "C:\Projekte\opc\.venv\Lib\site-packages\asyncua\common\structures104.py", line 299, in _generate_object
    exec(code, env)
  File "C:\Projekte\opc\.venv\Lib\site-packages\asyncua\common\structures104.py", line 474, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type, log_fail=log_ex)
  File "C:\Projekte\opc\.venv\Lib\site-packages\asyncua\client\client.py", line 833, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "C:\Projekte\opc\OPC_test\opc.py", line 106, in main
    await client.load_data_type_definitions()
  File "C:\Program Files\Python39\Lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Program Files\Python39\Lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Projekte\opc\OPC_test\opc.py", line 144, in <module> (Current frame)
    asyncio.run(main())
  File "C:\Program Files\Python39\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python39\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
TypeError: 'Field' object is not subscriptable
```

bauermarkus, 2023-09-15:
> depending on the settings in the s7 hw-config
> 
> you should try/except "await client.load_data_type_definitions()" if it fails its not the OPC UA 1.04 Structures, if an exception happens you need to "await client.load_type_definitions()" which should load the OPC UA 1.03 Structures client.py
> 
> ```python
>     async def load_type_definitions(self, nodes=None):
>         """
>         Load custom types (custom structures/extension objects) definition from server
>         Generate Python classes for custom structures/extension objects defined in server
>         These classes will available in ua module
>         WARNING: protocol has changed in 1.04. use load_data_type_definitions()
>         """
>         _logger.warning("Deprecated since spec 1.04, call load_data_type_definitions")
>         return await load_type_definitions(self, nodes)
> 
>     async def load_data_type_definitions(self, node: Optional[Node] = None, overwrite_existing: bool = False) -> Dict[str, Type]:
>         """
>         Load custom types (custom structures/extension objects) definition from server
>         Generate Python classes for custom structures/extension objects defined in server
>         These classes will be available in ua module
>         """
>         return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
> ```

Hello, thanks

A lot of types are loading correctly, it just breaks at some point

I expect it is 1.04
[image]

schroeder-, 2023-09-15:
Our import doesn't like a field named "List"
This snippet triggers the same error:
```
from typing import List
from dataclasses import dataclass
from dataclasses import field

@dataclass
class Test:
      List: List[int] = field(default_factory=[])
```

bauermarkus, 2023-09-15:
Okay, do you think it is fixable?

Is there a possibility so I only import certain areas?

schroeder-, 2023-09-15:
I am think about a solution. Manually importing is tricky because you needed to load them in the correct order, if they depend on each other.
Maybe i can share a monkey patch for quick help.

bauermarkus, 2023-09-15:
So the above structure is the most important for me, so if that is working, there is no quick fix needed from my side

Just in long term I guess it would be nice to not break :-)

schroeder-, 2023-09-18:
Here is a quick fix for you. This will monkey patch the problem.

```
from asyncua.common import structures104
import re
import keyword

def patched_clean_name(name):
    """
    Remove characters that might be present in  OPC UA structures
    but cannot be part of Python class names
    """
    if keyword.iskeyword(name):
        return name + "_"
    if name == 'List':
        return name + "_"
    if name.isidentifier():
        return name
    newname = re.sub(r'\W+', '_', name)
    newname = re.sub(r'^[0-9]+', r'_\g<0>', newname)
    _logger.warning("renamed %s to %s due to Python syntax", name, newname)
    return newname

structures104.clean_name = patched_clean_name
```

commit 36673c75 (2023-10-02): Allow List,Union,Optional named fields (#1454)

* Allow List,Union,Optional named fields

Fixes #1441

* use typing
closed 2023-10-02 by oroulet