FreeOpcUa/opcua-asyncio #1446: support for Aes256Sha256RsaPss security policy

antoine-galataud, 2023-09-19:
There doesn't seem to be an implementation for http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss yet.

I have started to implement one, see below standalone example that dynamically adds the new policy to `security_policies` module:

```python
import time

from asyncua.crypto import security_policies, uacrypto
from asyncua.crypto.security_policies import (
    Cryptography,
    DecryptorAesCbc,
    DecryptorRsa,
    EncryptorAesCbc,
    EncryptorRsa,
    MessageSecurityMode,
    SecurityPolicy,
    Signer,
    SignerHMac256,
    Verifier,
    VerifierHMac256,
    require_cryptography,
)
from cryptography.hazmat.primitives import hashes
from cryptography.hazmat.primitives.asymmetric import padding

def encrypt_rsa_oaep_sha256(public_key, data):
    ciphertext = public_key.encrypt(
        data,
        padding.OAEP(
            mgf=padding.MGF1(algorithm=hashes.SHA256()), algorithm=hashes.SHA256(), label=None
        ),
    )
    return ciphertext

def decrypt_rsa_oaep_sha256(private_key, data):
    text = private_key.decrypt(
        bytes(data),
        padding.OAEP(
            mgf=padding.MGF1(algorithm=hashes.SHA256()), algorithm=hashes.SHA256(), label=None
        ),
    )
    return text

def sign_pss_sha256(private_key, data):
    return private_key.sign(
        data,
        padding.PSS(
            mgf=padding.MGF1(algorithm=hashes.SHA256()), salt_length=padding.PSS.MAX_LENGTH
        ),
        hashes.SHA256(),
    )

def verify_pss_sha256(certificate, data, signature):
    certificate.public_key().verify(
        signature,
        data,
        padding.PSS(
            mgf=padding.MGF1(algorithm=hashes.SHA256()), salt_length=padding.PSS.MAX_LENGTH
        ),
        hashes.SHA256(),
    )

class SignerPssSha256(Signer):
    def __init__(self, client_pk):
        require_cryptography(self)
        self.client_pk = client_pk
        self.key_size = self.client_pk.key_size // 8

    def signature_size(self):
        return self.key_size

    def signature(self, data):
        return sign_pss_sha256(self.client_pk, data)

class VerifierPssSha256(Verifier):
    def __init__(self, server_cert):
        require_cryptography(self)
        self.server_cert = server_cert
        self.key_size = self.server_cert.public_key().key_size // 8

    def signature_size(self):
        return self.key_size

    def verify(self, data, signature):
        verify_pss_sha256(self.server_cert, data, signature)

class SecurityPolicyAes256Sha256RsaPss(SecurityPolicy):
    """Security policy Aes256_Sha256_RsaPss implementation.

    - SymmetricSignatureAlgorithm_HMAC-SHA2-256
      https://tools.ietf.org/html/rfc4634
    - SymmetricEncryptionAlgorithm_AES256-CBC
      http://www.w3.org/2001/04/xmlenc#aes256-cbc
    - AsymmetricSignatureAlgorithm_RSA-PSS-SHA2-256
      http://www.w3.org/2001/04/xmldsig-more#rsa-sha256
    - AsymmetricEncryptionAlgorithm_RSA-OAEP-SHA2-256
      http://www.w3.org/2001/04/xmlenc#rsa-oaep
    - KeyDerivationAlgorithm_P-SHA2-256
      http://docs.oasis-open.org/ws-sx/ws-secureconversation/200512/dk/p_sha256
    - CertificateSignatureAlgorithm_RSA-PKCS15-SHA2-256
      http://www.w3.org/2001/04/xmldsig-more#rsa-sha256
    - Aes256Sha256RsaPss_Limits
        -> DerivedSignatureKeyLength: 256 bits
        -> MinAsymmetricKeyLength: 2048 bits
        -> MaxAsymmetricKeyLength: 4096 bits
        -> SecureChannelNonceLength: 32 bytes
    """

    URI = "http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss"
    signature_key_size = 32
    symmetric_key_size = 32
    secure_channel_nonce_length = 32
    AsymmetricEncryptionURI = "http://opcfoundation.org/UA/security/rsa-oaep-sha2-256"
    AsymmetricSignatureURI = "http://opcfoundation.org/UA/security/rsa-pss-sha2-256"

    @staticmethod
    def encrypt_asymmetric(pubkey, data):
        return encrypt_rsa_oaep_sha256(pubkey, data)

    def __init__(self, peer_cert, host_cert, client_pk, mode, permission_ruleset=None):
        require_cryptography(self)
        if isinstance(peer_cert, bytes):
            peer_cert = uacrypto.x509_from_der(peer_cert)
        # even in Sign mode we need to asymmetrically encrypt secrets
        # transmitted in OpenSecureChannel. So SignAndEncrypt here
        self.asymmetric_cryptography = Cryptography(MessageSecurityMode.SignAndEncrypt)
        self.asymmetric_cryptography.Signer = SignerPssSha256(client_pk)
        self.asymmetric_cryptography.Verifier = VerifierPssSha256(peer_cert)
        self.asymmetric_cryptography.Encryptor = EncryptorRsa(
            peer_cert, encrypt_rsa_oaep_sha256, 66
        )
        self.asymmetric_cryptography.Decryptor = DecryptorRsa(
            client_pk, decrypt_rsa_oaep_sha256, 66
        )
        self.symmetric_cryptography = Cryptography(mode)
        self.Mode = mode
        self.peer_certificate = uacrypto.der_from_x509(peer_cert)
        self.host_certificate = uacrypto.der_from_x509(host_cert)
        if permission_ruleset is None:
            from asyncua.crypto.permission_rules import SimpleRoleRuleset

            permission_ruleset = SimpleRoleRuleset()

        self.permissions = permission_ruleset

    def make_local_symmetric_key(self, secret, seed):
        # specs part 6, 6.7.5
        key_sizes = (self.signature_key_size, self.symmetric_key_size, 32)

        (sigkey, key, init_vec) = uacrypto.p_sha256(secret, seed, key_sizes)
        self.symmetric_cryptography.Signer = SignerHMac256(sigkey)
        self.symmetric_cryptography.Encryptor = EncryptorAesCbc(key, init_vec)

    def make_remote_symmetric_key(self, secret, seed, lifetime):
        # specs part 6, 6.7.5
        key_sizes = (self.signature_key_size, self.symmetric_key_size, 32)

        (sigkey, key, init_vec) = uacrypto.p_sha256(secret, seed, key_sizes)
        if self.symmetric_cryptography.Verifier or self.symmetric_cryptography.Decryptor:
            self.symmetric_cryptography.Prev_Verifier = self.symmetric_cryptography.Verifier
            self.symmetric_cryptography.Prev_Decryptor = self.symmetric_cryptography.Decryptor
            self.symmetric_cryptography.prev_key_expiration = (
                self.symmetric_cryptography.key_expiration
            )

        # lifetime is in ms
        self.symmetric_cryptography.key_expiration = time.time() + (lifetime * 0.001)
        self.symmetric_cryptography.Verifier = VerifierHMac256(sigkey)
        self.symmetric_cryptography.Decryptor = DecryptorAesCbc(key, init_vec)

def define_policy_aes256_sha256_rsa_pss():
    setattr(
        security_policies, "SecurityPolicyAes256_Sha256_RsaPss", SecurityPolicyAes256Sha256RsaPss
    )
```

Let me know your thoughts. I can open a PR and integrate the code to the existing base if you feel it's worth it.

schroeder-, 2023-09-19:
I don't not an expert in secruity policies, but i does look good and if it works with other servers, than it's better than not supporting this security policy. 

Feel free to make a pr. Just add a test and theres is one place where you can configure the supported policies for the server, if you don't find it, I can look up the place.

antoine-galataud, 2023-09-19:
@schroeder- thanks for your inputs. I've opened #1447 and added some tests. I think I found where security policy lookup happens on server side, waiting for your review to confirm.
Note that there was a mistake in my first implementation posted in this issue, so I also edited it.
I'm also trying to validate the code with a proprietary server, this may require code changes.

antoine-galataud, 2023-09-19:
I managed to validate against a proprietary server. The only issue I spotted was with salt length (fixed to 32 bytes in the norm).

PR #1447 (2023-09-29): add support for Aes256Sha256RsaPss security policy
Fixes #1446
closed 2023-09-29 by oroulet