FreeOpcUa/opcua-asyncio #1448: Specify in install_requires that cryptography>=40.0.0 is required

apuester, 2023-09-20:
asyncio became incompatible with cryptography<40.0.0 with this commit: https://github.com/FreeOpcUa/opcua-asyncio/commit/97db2dd541d9c01dd5c4013f2faab059005e81ae

PrivateKeyTypes was added in cryptography 40.0.0 (see [here](https://cryptography.io/en/latest/hazmat/primitives/asymmetric/#cryptography.hazmat.primitives.asymmetric.types.PrivateKeyTypes))

Therefore this version constraint should be specified in install_requires of setup.py.

JohnnieFucker, 2023-09-21:
yes and it will conflict with pyopenssl when upgrade the cryptography to 40.0.0

pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pyopenssl 23.2.0 requires cryptography!=40.0.0,!=40.0.1,<42,>=38.0.0, but you have cryptography 40.0.0 which is incompatible.

linderd, 2023-11-28:
this problem still exists.   
setting pyopenssl to 23.2.0 and cryptography!=40.0.0,!=40.0.1,<42,>=38.0.0 is a good workaround, thx @JohnnieFucker

closed 2023-12-16 by oroulet

hansingt, 2024-01-08:
Thanks for fixing this. But the Version `1.0.5` needs to be yanked on pypi as well, because now, if for some reason, `cryptography<40` is required, pip will automatically install `asyncua==1.0.5`, because it does not contain the constraint and thus, the installation is broken:
```
> python -m venv venv
> ./venv/bin/pip install cryptography<40 asyncua
Collecting cryptography<40                                                                                                                                    
  Downloading cryptography-39.0.2-cp36-abi3-manylinux_2_28_x86_64.whl (4.2 MB)                                                                                
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.2/4.2 MB 8.6 MB/s eta 0:00:00                                                                                 
Collecting asyncua                                                                                                                                            
  Obtaining dependency information for asyncua from https://files.pythonhosted.org/packages/a8/5a/21eb5bc92de406d61b64ffc14c11a3a9d984f9fe9819fef3039d32ba0abf
/asyncua-1.0.6-py3-none-any.whl.metadata                                                                                                                      
  Using cached asyncua-1.0.6-py3-none-any.whl.metadata (9.3 kB)                                                                                               
Collecting cffi>=1.12 (from cryptography<40)                                                                                                                  
  Obtaining dependency information for cffi>=1.12 from https://files.pythonhosted.org/packages/9b/89/a31c81e36bbb793581d8bba4406a8aac4ba84b2559301c44eef81f4cf
5df/cffi-1.16.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata                                                                           
  Using cached cffi-1.16.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (1.5 kB)                                                       
Collecting aiofiles (from asyncua)                                                                                                                            
  Obtaining dependency information for aiofiles from https://files.pythonhosted.org/packages/c5/19/5af6804c4cc0fed83f47bff6e413a98a36618e7d40185cd36e69737f3b0e/aiofiles-23.2.1-py3-none-any.whl.metadata
  Using cached aiofiles-23.2.1-py3-none-any.whl.metadata (9.7 kB)
Collecting aiosqlite (from asyncua)
  Using cached aiosqlite-0.19.0-py3-none-any.whl (15 kB)
Collecting python-dateutil (from asyncua)
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pytz (from asyncua)
  Obtaining dependency information for pytz from https://files.pythonhosted.org/packages/32/4d/aaf7eff5deb402fd9a24a1449a8119f00d74ae9c2efa79f8ef9994261fc2/pytz-2023.3.post1-py2.py3-none-any.whl.metadata
  Using cached pytz-2023.3.post1-py2.py3-none-any.whl.metadata (22 kB)
INFO: pip is looking at multiple versions of asyncua to determine which version is compatible with other requirements. This could take a while.
Collecting asyncua
  Obtaining dependency information for asyncua from https://files.pythonhosted.org/packages/ff/0a/ed76899b4bf023d168bb5b62c558f141263f270f0e5e745c8a448f177b46/asyncua-1.0.5-py3-none-any.whl.metadata
  Using cached asyncua-1.0.5-py3-none-any.whl.metadata (9.3 kB)
Collecting sortedcontainers (from asyncua)
  Using cached sortedcontainers-2.4.0-py2.py3-none-any.whl (29 kB)
Collecting pyOpenSSL (from asyncua)
  Obtaining dependency information for pyOpenSSL from https://files.pythonhosted.org/packages/db/de/007b832ad7a95e6a73745609bbe123c407aa2c46bb0b8f765c8718294e7f/pyOpenSSL-23.3.0-py3-none-any.whl.metadata
  Using cached pyOpenSSL-23.3.0-py3-none-any.whl.metadata (12 kB)
Collecting typing-extensions (from asyncua)
  Obtaining dependency information for typing-extensions from https://files.pythonhosted.org/packages/b7/f4/6a90020cd2d93349b442bfcb657d0dc91eee65491600b2cb1d388bc98e6b/typing_extensions-4.9.0-py3-none-any.whl.metadata
  Using cached typing_extensions-4.9.0-py3-none-any.whl.metadata (3.0 kB)
Collecting pycparser (from cffi>=1.12->cryptography<40)
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
INFO: pip is looking at multiple versions of pyopenssl to determine which version is compatible with other requirements. This could take a while.
Collecting pyOpenSSL (from asyncua)
  Obtaining dependency information for pyOpenSSL from https://files.pythonhosted.org/packages/f0/e2/f8b4f1c67933a4907e52228241f4bd52169f3196b70af04403b29c6323
8a/pyOpenSSL-23.2.0-py3-none-any.whl.metadata
  Using cached pyOpenSSL-23.2.0-py3-none-any.whl.metadata (10 kB)
Collecting six>=1.5 (from python-dateutil->asyncua)
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Using cached asyncua-1.0.5-py3-none-any.whl (772 kB)
Using cached cffi-1.16.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (464 kB)
Using cached aiofiles-23.2.1-py3-none-any.whl (15 kB)
Downloading pyOpenSSL-23.2.0-py3-none-any.whl (59 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 59.0/59.0 kB 4.4 MB/s eta 0:00:00
Using cached pytz-2023.3.post1-py2.py3-none-any.whl (502 kB)
Using cached typing_extensions-4.9.0-py3-none-any.whl (32 kB)
[... cut ...]
  File "/home/tha/temp/venv/lib/python3.11/site-packages/asyncua/client/__init__.py", line 1, in <module>
    from .client import Client
  File "/home/tha/temp/venv/lib/python3.11/site-packages/asyncua/client/client.py", line 5, in <module>
    from cryptography.hazmat.primitives.asymmetric.types import PrivateKeyTypes
ImportError: cannot import name 'PrivateKeyTypes' from 'cryptography.hazmat.primitives.asymmetric.types' (/home/tha/temp/venv/lib/python3.11/site-packages/cryptography/hazmat/primitives/asymmetric/types.py)
```

PR #1507 (2023-12-16): bump cryptography to >40.0.1 in install_requires
closes #1448