FreeOpcUa/opcua-asyncio #1449: Using latest http://opcfoundation.org/UA/ nodeset 1.05.02

plamenk, 2023-09-21:
Dear community,

this is a copy / extended question from the wrongly created issue in https://github.com/FreeOpcUa/python-opcua/discussions/1525#discussioncomment-6861616. @schroeder- , many thanks for your hint. Indeed, I am using opcua-asyncio,can load many specs based on the included 1.04 spec and run my server, based on the examples you provided.
WOuld you be so kind and provide me with an example how to use the wrapper in order to import the file from https://reference.opcfoundation.org/api/nodesets/2.

Thanks in advance.

Plamen

schroeder-, 2023-09-22:
Your question is a bit unclear, an example for the sync layer -> https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/sync/server-example.py

If you just want to import a nodeset with the sync layer, I hope this snippet helps you here.
```
from asyncua import ua, uamethod
from asyncua.sync import Server, ThreadLoop

def main():
     with ThreadLoop() as tloop:
        # now set up our server
        server = Server(tloop=tloop)
        ....
        # Set up the server
        server.import_xml("custom_nodes.xml")
```

plamenk, 2023-09-23:
With the risk to misuse your patience, Alexander, here two tests - one, which I built based on the async server example and reduced to the essence, and another, I built based on your hint to the sync example. When I try to import the latest UA spec (1.05.02, included in the test), the following error appears in both cases:
asyncua.ua.uaerrors._auto.BadNodeIdExists: "The requested node id is already used by another node."(BadNodeIdExists)

Why I am trying to do this: the included "Machinery Basic Building Blocks_749922466.xml" requires the latest       <RequiredModel ModelUri="http://opcfoundation.org/UA/" Version="1.05.02" PublicationDate="2022-11-01T00:00:00Z" />

It appears that the internally used http://opcfoundation.org/UA/ is 1.05.01.

Thanks.

Plamen
[test_import_UA_1.05.02.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/12707383/test_import_UA_1.05.02.zip)

schroeder-, 2023-09-25:
First you can't import the the UA spec, because all nodes are already loaded.
There is a pull request that uses 1.05.02 internally https://github.com/FreeOpcUa/opcua-asyncio/pull/1444.
The other possiblity is to edit the spec Machinery Basic Building Blocks and target 1.05.01, this works most of the time.

plamenk, 2023-09-27:
Many thanks, @schroeder- . I adjusted the nodeset to use 1.05.01 and everything is working fine now.

francisjjp, 2023-10-05:
Hello.

Do you know if there is some version using pip install to update asyncua package with embedded OPC UA model version 1.05.02? maybe there is some date to put in "production" this new version that implements more recent OPC UA base model (1.05.02)?

Thanks in advance!

oroulet, 2023-10-05:
the MR is here: https://github.com/FreeOpcUa/opcua-asyncio/pull/1444
but it breaks some tests and I have no ideas of the consequences for fixing them.

SnoopAir, 2024-04-16:
> First you can't import the the UA spec, because all nodes are already loaded. There is a pull request that uses 1.05.02 internally #1444. The other possiblity is to edit the spec Machinery Basic Building Blocks and target 1.05.01, this works most of the time.

That first line was an eye opener. Could you point me to documentation which specifies it? I was "blindly" collecting all the dependencies of a custom NodeSet and always had trouble; some because of failing to import 1.0.5.02.
Is it possible to replace this "default" NodeSet by providing a path to another one?

GoetzGoerisch, 2024-04-16:
@SnoopAir  as @oroulet pointed out the UA 1.05.2 update for `asyncua` is pending. Most SDKs, `asyncua` included, don't allow an inplace update of the base spec.
Due to modelling tools the latest UA spec is used in companion specs. If the spec uses no new types defined, then one can manually modify the `<required models>` section in a CS.