FreeOpcUa/opcua-asyncio #1452: Simple client example causes RuntimeError: can't start new thread

pzcfoo, 2023-09-26:
**Describe the bug** <br />
I'm connecting to a server using minimal client example. The code has been modified slightly and anonymised. This code works against the minimal server example. When deployed a `RuntimeError: can't start new thread` is raised. This code is deployed as a docker container. The server is from Rockwell. Looking for guidance as I'm quite stuck on this one...

**To Reproduce**<br />
```
import asyncio
import logging
import os

import asyncua

logging.basicConfig()
_logger = logging.getLogger(__name__)
_logger.setLevel(logging.INFO)

async def test_opc():
    host = os.environ["OPC_HOST"]
    port = os.environ["OPC_PORT"]
    username = os.environ["OPC_USERNAME"]
    password = os.environ["OPC_PASSWORD"]
    url = f"opc.tcp://{host}:{port}/"  # no path is configured

    client = asyncua.Client(url=url)
    client.set_user(username)
    client.set_password(password)

    await client.connect()
    node = client.nodes.server_state
    node_value = await node.get_value()
    _logger.info(f"Node: {node_value}")
    await client.disconnect()

if __name__ == "__main__":
    asyncio.run(test_opc())
```

**Expected behavior**<br />
Is able to connect and log server_state information.

**Stacktrace**<br />
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/entry_point.py", line 34, in test_opc
    await client.connect()
  File "/usr/local/lib/python3.11/site-packages/asyncua/client/client.py", line 284, in connect
    await self.connect_socket()
  File "/usr/local/lib/python3.11/site-packages/asyncua/client/client.py", line 347, in connect_socket
    await self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)
  File "/usr/local/lib/python3.11/site-packages/asyncua/client/ua_client.py", line 300, in connect_socket
    await asyncio.wait_for(asyncio.get_running_loop().create_connection(self._make_protocol, host, port), self._timeout)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1045, in create_connection
    infos = await self._ensure_resolved(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 1419, in _ensure_resolved
    return await loop.getaddrinfo(host, port, family=family, type=type,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 867, in getaddrinfo
    return await self.run_in_executor(
                 ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 829, in run_in_executor
    executor.submit(func, *args), loop=self)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 176, in submit
    self._adjust_thread_count()
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 199, in _adjust_thread_count
    t.start()
  File "/usr/local/lib/python3.11/threading.py", line 957, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/entry_point.py", line 47, in <module>
    asyncio.run(test_opc())
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 189, in run
    with Runner(debug=debug) as runner:
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 63, in __exit__
    self.close()
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 73, in close
    loop.run_until_complete(loop.shutdown_default_executor())
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 571, in shutdown_default_executor
    thread.start()
  File "/usr/local/lib/python3.11/threading.py", line 957, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't start new thread

```

**Version**<br />
Python-Version: 3.11.5 <br />
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.4
docker version: 20.10.8

schroeder-, 2023-09-29:
This is not an issue of asyncua, check if your docker container limits threads.

closed 2023-09-29 by schroeder-

chrisjbremner, 2023-12-21:
I too encountered this issue after upgrading from version `0.9.92` to `1.0.4` of `asyncua`. I came across [this Stack Overflow post](https://stackoverflow.com/a/70093119) which pointed at [this blog](https://medium.com/nttlabs/ubuntu-21-10-and-fedora-35-do-not-work-on-docker-20-10-9-1cd439d9921), which suggested my Docker version was outdated. I checked my version (`Docker version 20.10.7, build f0df350`) and after updating (`Docker version 24.0.2, build cb74dfc`), I no longer see this error.