FreeOpcUa/opcua-asyncio #1455: Can't write real values

Jackarnd, 2023-09-29:
**Version**<br />
Python-Version:<br />
opcua-asyncio Version 1.0.4:

**Describe the bug** <br />
I have values defined in my PLC as `REAL`. 
I have no problem writing in `LREAL`s but when I try to write in a  `REAL` defined variable I get a mismatch error.
In my code I catch the exception to show the types like that :  
```
if "BadTypeMismatch" in str(e):
                print(f"Une erreur s'est produite : {str(e)}. Value to set is {type(value)} and node is {type(await self.__client.get_node(node_id).get_value())}")
```
This result in : `Value to set is <class 'float'> and node is <class 'float'>`.

Also I've just checked I have the same problem with an INT variable.

CodeSys defines `REAL` and `LREAL` as such :
<html>
<body>

Data Type | Lower Limit | Upper Limit | Memory
-- | -- | -- | --
REAL | -3.402823e+38 | 3.402823e+38 | 32 Bit
LREAL | -1.7976931348623158e+308 | 1.7976931348623158e+308 | 64 Bit

</body>
</html>

So I have tried with `numpy.float32()` but I get an another exception that just returns `'float32'`.

**To Reproduce**<br />
My function looks like that 

```
try:
            # Lire le nœud en utilisant son Node ID
            node = self.__client.get_node("somenodeaddress")
            # Écrire la nouvelle valeur dans le nœud
            await node.write_value(value)
            print(f"Value {value} written to Node {node_id}")
            return True
            
        except ua.uaerrors._auto.BadUserAccessDenied:
            print(f"Access denied: Node {node_id} is not writable.")
            return False
        except Exception as e:
                if "BadTypeMismatch" in str(e):
                    print(f"Une erreur s'est produite : {str(e)}. Value to set is {type(value)} and node is {type(await self.__client.get_node(node_id).get_value())}")
            else:
                print(f"Une erreur s'est produite : {str(e)}")
            return False
```

**Expected behavior**<br />

Can anyone tell me if there is a way to write in 32bits nodes ?

schroeder-, 2023-09-29:
Try this for 32 Bit:

`await node.write_value(ua.Float(value))`

For 64 Bit:
`await node.write_value(ua.Double(value))`

Jackarnd, 2023-09-29:
@schroeder-  Thank you it worked !

closed 2023-09-29 by Jackarnd