FreeOpcUa/opcua-asyncio #1469: StartTime and CurrentTime in ServerStatus node not update

impet14, 2023-10-16:
Hi, 

I found some thing in serverstatus node 2256 that StartTime and CurrentTime in the value of serverstatus node are not updating.
But the StartTime Node 2257 and CurrentTime node 2258 work correctly updating. 

[image]

Python-Version: 3.10
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.4

please suggest what I can do.

SebastianKarnappPTW, 2023-10-18:
Which server are you refering to?

KondratevAndrei, 2023-10-21:
```diff
@@ -72,6 +72,7 @@
             self, self.aspace, self.subscription_service, "Internal", user=User(role=UserRole.Admin)
         )
         self.current_time_node = Node(self.isession, ua.NodeId(ua.ObjectIds.Server_ServerStatus_CurrentTime))
+        self.serv_status_cur_time = Node(self.isession, ua.NodeId(ua.ObjectIds.Server_ServerStatus))
         self.time_task = None
         self._time_task_stop = False
         self.match_discovery_endpoint_url: bool = True
@@ -207,6 +208,13 @@
 
     async def _set_current_time_loop(self):
         while not self._time_task_stop:
+            try:
+                status = await self.serv_status_cur_time.get_value()
+                if status is not None:
+                    status.CurrentTime = datetime.utcnow()
+                    await self.serv_status_cur_time.write_value(status)
+            except Exception as e:
+                print(e)
             await self.current_time_node.write_value(datetime.utcnow())
             await asyncio.sleep(1)
```

impet14, 2023-10-30:
thanks so much .
so what is the good way please?

closed 2026-10-02 by oroulet