FreeOpcUa/opcua-asyncio #1473: Wrong Node type at ServerStatus Node

ikasakmongkon, 2023-10-20:
Hi all,

I found at the serverstatus node and its children node. 
all the node type are NumericNodeId but other stack such milo and open62541 are FourByteNodeId.

Also it seems the standard opc ua has define as FourByteNodeId.
https://reference.opcfoundation.org/Core/Part6/v105/docs/5.2.2.9#_Ref131423295

Thank you very for your consideration. 
I look forward to any response. 

here the result from my server.
result of freeopcua device
Node: Node(NumericNodeId(i=2257))
Node id: NumericNodeId(i=2257)
Browse name: QualifiedName(0:StartTime)
Display name: LocalizedText(Encoding:2, Locale:None, Text:StartTime)
Description: LocalizedText(Encoding:0, Locale:None, Text:None)
Value: 2023-10-17 03:10:06.778950

Can we define this all children node of Serverstatus node as Fourbytenodeid?

Python-Version: 3.10
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.4

schroeder-, 2023-10-20:
I dont see any issues here. Because NumericNodeId == FourByteNodeId. FourByteNodeId is only something that is used in the wireformat.

closed 2023-10-20 by schroeder-

impet14, 2023-10-23:
Hi @schroeder- 

thank you very much for your information.

I have test the opc ua stack with elipse mile and open62541
I have 3x opc ua servers using milo, freeopcua, and open 62541.
and create the freeopcua client to browse all server at serverstatus node. it worked.
But when I use milo opcua client to read all server, it can browse milo and open62541 but it cannot browse freeopcua server.

Result of freeopcua device.
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Java version: 11.0.20.1
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.6.11
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.6.11
[milo-nonce-util-secure-random] INFO org.eclipse.milo.opcua.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
freeopc Browse returned 0 references
BrowseResult(statusCode=StatusCode{name=Good, value=0x00000000, quality=good}, continuationPoint=ByteString{bytes=null}, references=[])

Result of milo device.
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Java version: 11.0.20.1
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.6.11
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.6.11
[milo-nonce-util-secure-random] INFO org.eclipse.milo.opcua.stack.core.util.NonceUtil - SecureRandom seeded in 0ms.
milo Browse returned 1 references
BrowseResult(statusCode=StatusCode{name=Good, value=0x00000000, quality=good}, continuationPoint=ByteString{bytes=null}, references=[ReferenceDescription(referenceTypeId=NodeId{ns=0, id=40}, isForward=true, nodeId=ExpandedNodeId{ns=0, id=2138, serverIndex=0}, browseName=QualifiedName{name=ServerStatusType, namespaceIndex=0}, displayName=LocalizedText{text=ServerStatusType, locale=en}, nodeClass=VariableType, typeDefinition=ExpandedNodeId{ns=0, id=0, serverIndex=0})])

Result of open62541 device.
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Java version: 11.0.20.1
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Stack version: 0.6.11
[main] INFO org.eclipse.milo.opcua.sdk.client.OpcUaClient - Eclipse Milo OPC UA Client SDK version: 0.6.11
[milo-nonce-util-secure-random] INFO org.eclipse.milo.opcua.stack.core.util.NonceUtil - SecureRandom seeded in 2ms.
open62541 Browse returned 1 references
BrowseResult(statusCode=StatusCode{name=Good, value=0x00000000, quality=good}, continuationPoint=ByteString{bytes=null}, references=[ReferenceDescription(referenceTypeId=NodeId{ns=0, id=40}, isForward=true, nodeId=ExpandedNodeId{ns=0, id=2138, serverIndex=0}, browseName=QualifiedName{name=ServerStatusType, namespaceIndex=0}, displayName=LocalizedText{text=ServerStatusType, locale=}, nodeClass=VariableType, typeDefinition=ExpandedNodeId{ns=0, id=0, serverIndex=0})])

I notice something when I use freeopcua to browse the ServerStatus Node 2256. I could read all from freeopcua/milo/open62541 devices.
but it seem the node type is difference. from freeopcua result is NumericNodeId but from milo/open62541 is FourByteNodeId
result of freeopcua device
Node: Node(NumericNodeId(i=2257))
Node id: NumericNodeId(i=2257)
Browse name: QualifiedName(0:StartTime)
Display name: LocalizedText(Encoding:2, Locale:None, Text:StartTime)
Description: LocalizedText(Encoding:0, Locale:None, Text:None)
Value: 2023-10-17 03:10:06.778950

result of milo device
Requested session timeout to be 3600000ms, got 120000ms instead
Node: Node(FourByteNodeId(i=2257))
Node id: FourByteNodeId(i=2257)
Browse name: QualifiedName(0:StartTime)
Display name: LocalizedText(Encoding:3, Locale:en, Text:StartTime)
Description: LocalizedText(Encoding:0, Locale:None, Text:None)
Value: 2023-10-19 02:29:48.174000

result of open62541 device
Node: Node(FourByteNodeId(i=2257))
Node id: FourByteNodeId(i=2257)
Browse name: QualifiedName(0:StartTime)
Display name: LocalizedText(Encoding:3, Locale:, Text:StartTime)
Description: LocalizedText(Encoding:0, Locale:None, Text:None)
Value: 1970-01-01 00:00:00.255000

impet14, 2023-10-24:
The following code is for milo client to browse the servers that I use.

```
Here is a variation that will compile and run with 0.6.11:

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.stack.core.Identifiers;
import org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseDirection;
import org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseResultMask;
import org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass;
import org.eclipse.milo.opcua.stack.core.types.structured.BrowseDescription;
import org.eclipse.milo.opcua.stack.core.types.structured.BrowseResult;

import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;

class Scratch {
    public static void main(String[] args) throws Exception {
        var client = OpcUaClient.create("opc.tcp://localhost:4840");
        client.connect().get();

        BrowseResult result = client.browse(new BrowseDescription(
            Identifiers.Server_ServerStatus,
            BrowseDirection.Forward,
            Identifiers.HasTypeDefinition,
            false,
            uint(NodeClass.ObjectType.getValue() | NodeClass.VariableType.getValue()),
            uint(BrowseResultMask.All.getValue())
        )).get();

        System.out.println("Browse returned " + result.getReferences().length + " references");
    }
}
```
Here is the code of freeopcu client, I use to browse the node.

```
# import the freeopcua library
import opcua
from opcua import ua

# create a client object
client = opcua.Client("opc.tcp://localhost:4840")
# client = opcua.Client("opc.tcp://localhost:12686/milo")

# define a recursive function to print all children of a node
def print_all_children(node):
    # print the attributes and values of the node
    print(f"Node: {node}")
    print(f"Node id: {node.nodeid}")
    print(f"Browse name: {node.get_browse_name()}")
    print(f"Display name: {node.get_display_name()}")
    print(f"Description: {node.get_description()}")
    try :
        print(f"Value: {node.get_value()}")
    except opcua.ua.uaerrors.UaError as e:
        # handle any OPC UA related error
        print(f"An OPC UA error occurred: {e}")
    except Exception as e:
        # handle any other generic error
        print(f"An unexpected error occurred: {e}")
    print()

    # get the children of the node
    children = node.get_children()

    # if the node has no children, return
    if not children:
        return

    # otherwise, call the function on each child
    for child in children:
        print_all_children(child)

# define a function to browse the specific node
def browse_specific_node():
    # connect to the server
    client.connect()

    # specify the desired namespace index and node id
    ns_index = 0
    node_id = 2256

    # get the specific node by its node id
    specific_node = client.get_node(ua.NodeId(node_id, ns_index))

    # call the recursive function on the specific node
    print_all_children(specific_node)

    # disconnect from the server
    client.disconnect()

# run the function
browse_specific_node()

```

schroeder-, 2023-10-24:
The reference handling of our brows is a bit limited. 
try `Identifiers.HasComponent` instead of `Identifiers.HasTypeDefinition`

impet14, 2023-10-24:
```
BrowseResult result = client.browse(new BrowseDescription(
            Identifiers.Server_ServerStatus,
            BrowseDirection.Forward,
            // Identifiers.HasTypeDefinition,
            Identifiers.HasComponent,
            false,
            uint(NodeClass.ObjectType.getValue() | NodeClass.VariableType.getValue()),
            uint(BrowseResultMask.All.getValue())
        )).get();

        System.out.println("freeopcua Browse returned " + result.getReferences().length + " references\r\n" + result.toString());
       ```
        the reference also got 0.
thank you for your suggestion. Is there any other way please?

schroeder-, 2023-10-24:
This should work
```

        BrowseResult result = client.browse(new BrowseDescription(
            Identifiers.Server_ServerStatus,
            BrowseDirection.Forward,
            Identifiers.HasComponent,
            false,
            uint(NodeClass.Object.getValue() | NodeClass.Variable.getValue()),
            uint(BrowseResultMask.All.getValue())
        )).get();
```

impet14, 2023-10-24:
```
BrowseResult result = client.browse(new BrowseDescription(
            Identifiers.Server_ServerStatus,
            BrowseDirection.Forward,
            Identifiers.HasComponent,
            false,
            uint(NodeClass.Object.getValue() | NodeClass.Variable.getValue()),
            uint(BrowseResultMask.All.getValue())
        )).get();

        System.out.println("freeopc Browse returned " + result.getReferences().length + " references\r\n" + result.toString());

```

Nice! thank you so much. I worked but got 6 references. anyway thanks so much for your time. 
I appreciate your kindness.