FreeOpcUa/opcua-asyncio #1474: Python OPC UA call method with an KeyValusPaia argument as input

aikawa1981, 2023-10-22:
Hi. I am new to OPC UA and stuck with calling a method with an KeyValuePair argument as input. 
I posted the details of my question on 
[Stack Overflow]
(https://stackoverflow.com/questions/77338446/call-method-with-keyvaluepair-input-argument) 
Can anybody give me a hint? Thanks!

AndreasHeine, 2023-10-22:
https://github.com/AndreasHeine/opcua-tutorial/blob/main/client/call_method.py

aikawa1981, 2023-10-22:
I found the server's log file and the command sent by Uaexpert is as follows:
`22.10.2023 08:00:17.006 <6> NCOPCUAINFO #CALL_METHOD 
2:UpdateToolDataItems #SESSION ns=3;i=3930534315 #OBJECT ns=1;i=58060 
#ARG in:| Value(KeyValuePairArray) = Key: 1:Length, Value:  Value(Double) = 258.256; /;
| Value(UInt32) = 10;| Value(UInt8) = 0;|`

I have tried Variant of type ExtensionObject as follows:
[image]

AndreasHeine, 2023-10-22:
value = ua.KeyValuePair()
arg = ua.Variant(value, ua.VariantType.ExtebsionObject)

aikawa1981, 2023-10-22:
I've tried again as follows,I got BadNoMatch
 arg  data format is the same as the server output
[image]

Data format output from the server as follows :
`Variant(Value=KeyValuePair(Key=QualifiedName(NamespaceIndex=1, Name='Length'), Value=Variant(Value=150.0, VariantType=<VariantType.Double: 11>, Dimensions=None, is_array=False)), VariantType=<VariantType.ExtensionObject: 22>, Dimensions=None, is_array=False)`

AndreasHeine, 2023-10-22:
your error happens not on methodcall. it happens before!?

aikawa1981, 2023-10-22:
sorry! I defined "UpdateToolDataItems" twice
I fixed the code and the problem was BadInvaildArguments
[image]

AndreasHeine, 2023-10-23:
have you checkt the order of args and the correctness of your varianttypes?

aikawa1981, 2023-10-30:
Finally I tried this：
arg1 = ("1:Length")
    arg2 = ua.Variant(258, ua.VariantType.Double)
    arg3 = [ua.KeyValuePair(arg1,arg2)]
    update_tool_data_method = await tool_data_methods.get_child([f"{id_types}:UpdateToolDataItems"])
    update_data = await tool_data_methods.call_method(
        update_tool_data_method,
        arg3,
        ua.UInt32(12),
        ua.Byte(0)
    )
  It works well
Thanks!