FreeOpcUa/opcua-asyncio #1477: API Question: How to get the original OPCUA Error as client

f0rw4rd, 2023-10-26:
Hi!

I am working on a OPC UA client and need to check what is the reason why the connection failed e.g. BadTcpEndpointUrlInvalid. So far I have found no information or any hint how to get the original OPC UA error code and a TimeoutError is the only returned exception at the moment. Sorry if this is duplicate, but can you give me a pointer in the right direction ?  

Thanks

SebastianKarnappPTW, 2023-11-15:
Can you share your code, where you're running into a problem? Is it with the Client?

oroulet, 2023-11-15:
When the server answers BadTcpEndpointUrlInvalid then tht exact exception is raised. IF you get a TimeoutError, then it means the server did not answer in time...

closed 2025-12-28 by f0rw4rd