FreeOpcUa/opcua-asyncio #1479: When running the example statemachine-example.py, an exception occurs.

ernestsheu, 2023-10-31:
When running the example statemachine-example.py, an exception occurs.

INFO:asyncua.common.instantiate_util:Instantiate: Skip node without modelling rule QualifiedName(NamespaceIndex=0, Name='InitialStateType') as part of QualifiedName(NamespaceIndex=2, Name='Finished')
INFO:asyncua.common.instantiate_util:Instantiate: Skip node without modelling rule QualifiedName(NamespaceIndex=0, Name='ChoiceStateType') as part of QualifiedName(NamespaceIndex=2, Name='Finished')
WARNING:asyncua.server.address_space:Write refused: Variant: Variant(Value=NodeId(Identifier=6, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>), VariantType=<VariantType.NodeId: 17>, Dimensions=None, is_array=False) with type VariantType.NodeId does not have expected type: VariantType.Null
Traceback (most recent call last):
  File "statemachine-example.py", line 87, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "statemachine-example.py", line 54, in main
    await mystatemachine.change_state(state1, trans1, f"{mystatemachine._name}: Idle", 300)
  File "/home/abc/.local/lib/python3.8/site-packages/asyncua/common/statemachine.py", line 183, in change_state
    await self._write_state(state)
  File "/home/abc/.local/lib/python3.8/site-packages/asyncua/common/statemachine.py", line 206, in _write_state
    await self._current_state_id_node.write_value(state.node.nodeid, varianttype=ua.VariantType.NodeId)
  File "/home/abc/.local/lib/python3.8/site-packages/asyncua/common/node.py", line 255, in write_value
    await self.write_attribute(ua.AttributeIds.Value, dv)
  File "/home/abc/.local/lib/python3.8/site-packages/asyncua/common/node.py", line 307, in write_attribute
    result[0].check()
  File "/home/abc/.local/lib/python3.8/site-packages/asyncua/ua/uatypes.py", line 373, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadTypeMismatch: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)

PKatus, 2023-12-13:
I can confirm the problem. Neither statemachine-example.py nor statemachine-example.py finitestatemachine-example.py seems to work.
Although I am still struggling to understand how OPC works at all, I assume the problem is a missmatch in the type definitions.

There was a fix BadTypeMismatch in statemachine.py #1334 (Fixed by #1335), which changed the write of the "Id" to ua.VariantType.NodeId. But the type of "Id" seems to be ua.ObjectIds.String in the actual version. 

(see opcua-asyncio/asyncua/server/standard_address_space/standard_address_space_services.py)

The DataType of the ID variable should be BaseDataType according to 
https://reference.opcfoundation.org/Core/Part16/v105/docs/4.4

For my understanding this means that it could have any subtype belongoing to Basetype. For some reason ua.ObjectIds.String was chosen. Wouldn't a number (the states) be more useful?

Anyway:

The statemachine-example should run when you change two lines of code in opcua-asyncio/asyncua/common
/statemachine.py:

I have changed the line 206 in _write_state
`await self._current_state_id_node.write_value(state.node.nodeid, varianttype=ua.VariantType.NodeId)`

and line 225 in _write_transition

`await self._last_transition_id_node.write_value(transition.node.nodeid, varianttype=ua.VariantType.NodeId)`

to

`await self._current_state_id_node.write_value(state.name,ua.VariantType.String)`
and 
`await self._last_transition_id_node.write_value(transition.name, ua.VariantType.String)`

tmcguc, 2024-04-22:
The fix #1335 worked.  doesn't seem like it is the version in pip how do we patch?