FreeOpcUa/opcua-asyncio #1493: [Question] uamethod no parent

AiyionPrime, 2023-11-13:
Good afternoon everyone;
I came across uamethod these days and wnted to know wheter it would be possible to use it as decorator for functions whose signature does not have "parent" as first attribute?

When the fuction decorated with it is actually a method and the class is already the parent of the "function" I see little use in having it being part of every of our server methods?

If I understand it correctly a method currently can't look like this, if I want to decorate it with uamethod, even if it' not making use of 'parent'?

```python
class PinController:

    def on_set_output(self, pin: int, value: bool) -> None:
        # set the pin to something useful
        pass
```

AndreasHeine, 2023-11-13:
ua methods are always part of an opc ua objecttype so the always have a parent!

AndreasHeine, 2023-11-13:
you probably need to create a own Pincontrollertype, instantiate a instance of that type and bind the instances methods to a class function

AiyionPrime, 2023-11-13:
Maybe my example was misleading? If I were to implement an OPC UA Server based on PinController, and wanted the method `on_set_output` to be the OPC UA "function" that handles the request, then it is nice, that the server gives a parent attribute in it's direction, but I don't want it.

Lets see. The Class above could have a reference to a server like this and link methods accordingly.

I've updated the example above; maybe my case became somewhat clearer. Sorry for the confusion.

schroeder-, 2023-11-17:
Just make a wrapper function

```
 def remove_parent(func):
     def wrap(parent, *args):
          return func(*args)
     return wrap

class PinController:

    def __init__(self, some_server: Server):
        self.server = some_server
        server.link_method(
            server.get_node(ua.NodeId.from_string("ns=1;i=42")), uamethod(remove_parent(self.on_set_output))
        )

    def on_set_output(self, pin: ua.UInt32, value: ua.Boolean) -> None:
        # set the pin to something useful
        pass
```