FreeOpcUa/opcua-asyncio #1506: Cannot read node for format "nsu=xxxxx;s=yyyyy"

xky0007, 2023-11-28:
**Describe the bug** <br />
Cannot read node for node id format "nsu=xxxxx;s=yyyyy"

**To Reproduce**<br />
```
        id = "nsu=http://xyz.com/UA/Connectivity/;s=35.Running"
        node = client.get_node(id)
        value = await node.read_value()
```
Then, get error `asyncua.ua.uaerrors._auto.BadEncodingLimitsExceeded: The message encoding/decoding limits imposed by the stack have been exceeded.(BadEncodingLimitsExceeded)`

**Expected behavior**<br />
Read the value.

**Version**<br />
Python-Version: 3.10.12 <br />
opcua-asyncio Version (e.g. master branch, 0.9): 0.9

schroeder-, 2023-11-28:
First does it work if you don't use nsu=xxx? Because the error doesn't look like it is caused by the nodeid. More like the value you try to read is to long for the network settings. Do you have a full backtrace? Also a wireshark trace would be helpful.

xky0007, 2023-12-04:
> Ok your use case is not supported via nsu, currently.
> 
> You could rewrite the code like this, then it should work:
> 
> ```
> idx = await client.get_namespace_index('http://xyz.com/UA/Connectivity/')
> node = client.get_node(ua.NodeId('35.Running', idx))
> ```

Thanks. It works now.

Very thanks for your help.

closed 2023-12-04 by xky0007