FreeOpcUa/opcua-asyncio #1509: change the init node source timestamp using Subscription

xyzakdkd, 2023-11-29:
## Set the initial source timestamp for all nodes using a subscription to the current time when the code is executed

I'm new to opcua, and I'm facing an issue when running my script. Upon retrieving all nodes, I observe that the subscription mode returns outdated data, possibly from a days ago. because some sensors do not change when starting the process. how can I modify the initial values to get the timestamp of running the script and than do the subscription.

```
    def datachange_notification(self, node: Node, val, data):
        node = str(node)
        if node in SubHandler.result.keys():
            source_timestamp = data.monitored_item.Value.SourceTimestamp
            server_timestamp = data.monitored_item.Value.ServerTimestamp
            print(f"source timestamp : {source_timestamp}, Value :  {SubHandler.result[node][0]}  =  {val}")
        else:
            print("Node not found")

```

**Expected behavior**<br />
In my current implementation, I've noticed that when obtaining to get the initial values of all nodes, such as the 'position_z' node, the timestamp appears to be outdated. first I aim to retrieve all nodes from my node dictionary with timestamps set to when the script is executed and than work like a normal subscription and get the source timestamp in the subscription.

**Output**<br />
[image]

schroeder-, 2023-11-29:
This is not a issue. Because this is as expected. The value timestamp is only updated if the value has changed.

xyzakdkd, 2023-11-29:
ok thanks, and how can i achieve what i want

schroeder-, 2023-11-30:
```
class SubHandler:
   def __init__(self):
       self.start_up = datetime.utcnow()
   .... 
   def datachange_notification(self, node: Node, val, data):
        node = str(node)
        if node in SubHandler.result.keys():
            source_timestamp = data.monitored_item.Value.SourceTimestamp
            server_timestamp = data.monitored_item.Value.ServerTimestamp
            if source_timestamp > self.startup:
                source_timestamp = self.startup
             if server_timestamp> self.startup:
                server_timestamp = self.startup
            print(f"source timestamp : {source_timestamp}, Value :  {SubHandler.result[node][0]}  =  {val}")
        else:
            print("Node not found")
```

schroeder-, 2023-12-01:
a client cant change the timestamp on the server. So you have todo this on your client code as my example does.

xyzakdkd, 2023-12-06:
def __init__(self):
        self.start_up = datetime.utcnow()
        self.first_call = True
         ...
     def datachange_notification(self, node: Node, val, data):
            node = str(node)
            if node in SubHandler.result.keys():
                source_timestamp = data.monitored_item.Value.SourceTimestamp
                server_timestamp = data.monitored_item.Value.ServerTimestamp

                if source_timestamp < self.start_up:
                    dict_init_values = {"object": SubHandler.result[node][1], "Source timestamp" : str(self.start_up), "node": SubHandler.result[node][0] ,"value": val}
                    key = SubHandler.result[node][0]+SubHandler.result[node][1]
                    if not SubHandler.init_data or val in SubHandler.init_data:
                         SubHandler.init_data[key] = dict_init_values
                    else:
                         SubHandler.init_data[key] = dict_init_values 

                if source_timestamp > self.start_up:
                    if self.first_call:
                         for key, value in SubHandler.init_data.items():
                            print(value)
                            self.first_call =False
                    print(f"source timestamp : {source_timestamp} , value :  {SubHandler.result[node][0]}  =  {val}, object: {SubHandler.result[node][1]}")        
            else:
                print("Node not found")

i store the init nodes and values on a dictionary ( that way i will get all the news values before the source timestamp start) to get the last value before the source timestamp is greater than the utc but sometimes i dont get all the nodes is the dictionary because the source timestamp. can you please help.