FreeOpcUa/opcua-asyncio #1510: Writing variables to ABB IRC5 controller using set_value() function

Engeenur, 2023-11-30:
I am having problems with using the node.set_value() function. I keep getting a "The value supplied for the attribute is not of the same type as the attribute"s value."(BadTypeMismatch) error. I can change the variable value with the Softing OPC UA Client software.

I tried all sorts of things:
dv = 1.0
self.start_flag_node.set_value(dv)

dv = opcua.ua.DataValue(opcua.ua.Variant(3.0, opcua.ua.VariantType.Double))
dv.ServerTimestamp = None
dv.SourceTimestamp = None
self.start_flag_node.set_value(dv)

dv = opcua.ua.Variant(3.0, opcua.ua.VariantType.Double)
self.start_flag_node.set_value(dv)

and probably some others.

No matter what I do I still get the same error. The DataType of variable that the Softing OPC UA Client detects this node to have is a Double (i=11).

schroeder-, 2023-11-30:
Are you using asyncua or python opcua? Because from the module names it looks you are using the old library.

If you are using python opcua try first to switch to asyncua with either async oder use the syncwrapper.

schroeder-, 2023-11-30:
Also this is the easiest solution:

```
dv = 1.0
self.start_flag_node.set_value(ua.Double(dv))

```