FreeOpcUa/opcua-asyncio #1516: OPCUA Subscription limit

shanmuk-quartic, 2023-12-05:
On a server with around 6000 tags streaming with frequency of 1 second, we are seeing lag build up. What are the expected throughput for python opcua subscriptions. On 3000 tags, it is working fine

AndreasHeine, 2023-12-05:
depends entirely on your CPU / Ram

shanmuk-quartic, 2023-12-05:
I have 4 vcpu / 16 GB machine and Python version 3.8. This configuration works great for 3000 tags. At 6000 tags, This is creating lag

shanmuk-quartic, 2023-12-05:
Thanks for the fast reply. This is with a single process setup and using all tags in single subscription

shanmuk-quartic, 2023-12-05:
PublishInterval = 1 second
sampling_interval = 500 millisecond
queuesize = 50
no filter as such all updates are received
I agree with the interpreter part. But just thinking if it is possible to scale by settings or by adding multiprocessing.

AndreasHeine, 2023-12-05:
scaling is working by using multiple clientconnections/instances sharing the load of nodes to monitor! opcua-asyncio is using an EventLoop which makes multiprocessing... how to say politely ^^ Not a path you want to go...

guess your using Linux which has another large influence on the EventLoop performance!?

shanmuk-quartic, 2023-12-05:
I have another process that pushes to remote kafka all received datapoints. Just to note if i use 8 vcpu, it is again working like a charm with kafka receiving at below 3 second latency at 6000 datapoints

shanmuk-quartic, 2023-12-05:
If i were to split the load in 2 two processes into 3000 tags each, it is consuming too much cpu causing delays in other processing. Is that supposed to be the case?

AndreasHeine, 2023-12-05:
actually it should enable the OS to run the two Python-Processes (a Client-Instance each) on different CPU Cores (tbh not sure how vCores work in that regard)

shanmuk-quartic, 2023-12-05:
Good point. I too was thinking along the same lines and expected performance improvement because of 4 vcores. Unexpected cpu rise was the case though. Will debug it more on why that was the case. Is there any testing metrics i can go with to see whether my code is doing fine for single subscription with 6000 tags? 

https://github.com/FreeOpcUa/python-opcua/issues/701#issuecomment-424476282
Is such testing done by any chance?

AndreasHeine, 2023-12-05:
usually the server exposes their limits how many monitoreditems it supports per subscription...

https://reference.opcfoundation.org/Core/Part5/v104/docs/6.3.11
its part of the server object -> ServerCapabilities -> OperationalLimits

but just a numbers game -> if you have like 5000 msg/s you have 5ms time per message this is a lot for python as long as you do just pushing towards kafka i would use a async generator (tends to have the best performance) to fill the gap between us-client and kafka

shanmuk-quartic, 2023-12-05:
Got it thanks

closed 2023-12-05 by shanmuk-quartic