FreeOpcUa/opcua-asyncio #1520: BadWriteNotSupported and TypeError: must be called with a dataclass type or instance

maxdd, 2023-12-10:
Hello,
this is not really a bug but more an issue i'm failing to solve.

My use case is a PLC whose variables for whatever reason (might) have "timestamps" (Source and Server) either locked or basically not  writeable. So whenever i do 

`await h_tmax_m2.write_value(60)`

i receive 

`ERROR: The server does not support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)`

I was hoping to create a Datavalue with `SourceTimestamp=None and ServerTimestamp=None` mainly because another program i have is not complaining about it when while is uses these values although i'm not sure whether it will fix the issue.
The write operation works but the program crashes and i didnt want to wrap everything in a try except caluse
So i've tried something like

```
dv = ua.DataValue(
 ua.Variant(
      100, 
      ua.VariantType.Int16
  ),
  StatusCode_=ua.StatusCodes.Good,
  SourceTimestamp=None,
  ServerTimestamp=None
)
await h_tmax_m2.write_attribute(ua.AttributeIds.Value, dv)
```
but i receive the following 

`TypeError: must be called with a dataclass type or instance`

which i'm struggling to understand why.
What am i doing wrong in the cose above? Is there a better way to solve this?

Regards,

AndreasHeine, 2023-12-10:
plc type abd vendor?

maxdd, 2023-12-10:
It should be a Schneider M241

AndreasHeine, 2023-12-10:
try to create the datavalue without servertimestamp!? the servertimestamp is usually the point in time in which the server recv that datapoint!

you can pass the dv to the write_value method!

maxdd, 2023-12-10:
Hello,
that solved the issue

```
dv = ua.DataValue(
          ua.Variant(
              100, 
              ua.VariantType.Int16
          )
        )
await h_tmax_m2.write_value(dv)
```
I have one more question, my PLC is roughly 3 mins behind in terms of time right now. It seems like it has a drift.
Given how powerful and useful this library is i was wondering whether i could fix that.
What is the proper method to fix it? I do see there is a Status/CurrentTime variable which is not writeable. Shall i ask for a permission change or is there a more adequate method?

AndreasHeine, 2023-12-10:
most plc dont have battery buffered RTC.... so a configuring a NTP-Server is the best option

closed 2023-12-10 by maxdd