FreeOpcUa/opcua-asyncio #1524: Reading Custom Structure with B&R OPC UA client fails with Error 0x80390000 Bad_DataEncodingUnsupported

MaCar37, 2023-12-13:
**Describe the bug** <br />
I can not read custom structures from an opcua-asyncio Server using a B&R PLC with an OPC UA client. Reading always fails with Error 0x80390000 Bad_DataEncodingUnsupported.

I already contacted the B&R Suppport. B&R claims the Error indicates that the Sever uses an old implementation of the OPCUA standard.

I also tried other OPC UA Servers where my B&R client could read custom structures.

The same problem might be already mentioned inside the Discussion of the older librabry python-opcua here: [#1527](https://github.com/FreeOpcUa/python-opcua/discussions/1527)

**To Reproduce**<br />
I used the example server-custom-structures-and-enums.py from this repository.
For the client I used the OpcUa_Sample from B&R Automation Studio V4.9.6.42.

**Version**<br />
Python-Version:<br /> Python 3.9.7
opcua-asyncio Version (e.g. master branch, 0.9): <br /> asyncua 1.0.5

AndreasHeine, 2024-02-11:
what encoding is it?

there where new (1.04) and old (1.03) structures! 
have you tried both:
```python
    async def load_type_definitions(self, nodes=None):
        """
        Load custom types (custom structures/extension objects) definition from server
        Generate Python classes for custom structures/extension objects defined in server
        These classes will available in ua module
        WARNING: protocol has changed in 1.04. use load_data_type_definitions()
        """
        _logger.warning("Deprecated since spec 1.04, call load_data_type_definitions")
        return await load_type_definitions(self, nodes)

    async def load_data_type_definitions(self, node: Optional[Node] = None, overwrite_existing: bool = False) -> Dict[str, Type]:
        """
        Load custom types (custom structures/extension objects) definition from server
        Generate Python classes for custom structures/extension objects defined in server
        These classes will be available in ua module
        """
        return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
```

rempAut, 2024-02-11:
To specify my use case, I have a server method returning a custom structure.

I tried it with the` load_data_type_definitions()` first, and with uaExpert client it worked. Didn't work in B&R client.
Next I tried with `load_type_definitions()` and neither uaExpert and B&R client worked.

rempAut, 2024-02-11:
So for now I switched from `load_data_type_definitions()` to the old `load_type_definitions()` and I am able to get a valid call in uaExpert again. That means in uaExpert, both loading procedures are working.

However, more B&R problems appeared, because I am no longer to get a method handle on the PLC side. Calling `UA_MethodGetHandle` returns error code 0x8074_0000.  This B&R library just stinks.

rempAut, 2024-02-16:
Allright, so I got it running, but I don't completely understand why.

* I used the latests Automation Runtime I4.93
* changed the OPCUA system configuration to use the 1.04 structure structure definitions
[image]
* declared the structures in Automation Studio
* used `server.import_xml()` to import the generated nodeset file with stucture definitions from Automation Studio.

Now here is the part where I would like to ask for some clarification. I was investigating whatever I could find about the `Bad_DataEncodingUnsupported`. I noticed a difference in nodesets exported from opcua-asyncio and Automation Studio in the encoding nodes (`UAObject` vs `UAObjectType`) Is this relevant? I am attaching snippets from the nodeset files.

## Nodeset snippet - exported from Automation Studio
```
   <UADataType NodeId="ns=1;i=101"
                BrowseName="1:parentElementType">
        <DisplayName>parentElementType</DisplayName>
        <Description>parentElementType</Description>
        <References>
            <Reference ReferenceType="HasSubtype"
                       IsForward="false">i=22</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=102</Reference>
        </References>
        <Definition Name="1:parentElementType">
            <Field DataType="i=12"
                   Name="ID" />
            <Field DataType="i=6"
                   Name="Length" />
            <Field DataType="i=6"
                   Name="Diameter" />
        </Definition>
    </UADataType>
    <UAObject SymbolicName="DefaultBinary"
              NodeId="ns=1;i=102"
              BrowseName="Default Binary">
        <DisplayName>Default Binary</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
            <Reference ReferenceType="HasEncoding"
                       IsForward="false">ns=1;i=101</Reference>
        </References>
    </UAObject>
```

##  Nodeset snippet - python-asyncua `server.export()` 
```
    <UADataType NodeId="ns=1;i=3"
                BrowseName="1:parentElementType">
        <DisplayName>parentElementType</DisplayName>
        <Description>parentElementType</Description>
        <References>
            <Reference ReferenceType="HasSubtype"
                       IsForward="false">i=22</Reference>
            <Reference ReferenceType="HasEncoding">ns=1;i=4</Reference>
        </References>
        <Definition Name="parentElementType">
            <Field DataType="i=12"
                   Name="ID" />
            <Field DataType="i=6"
                   Name="Length" />
            <Field DataType="i=6"
                   Name="Diameter" />
        </Definition>
    </UADataType>
    <UAObjectType NodeId="ns=1;i=4"
                  BrowseName="0:DefaultBinary">
        <DisplayName>DefaultBinary</DisplayName>
        <Description>DefaultBinary</Description>
        <References>
            <Reference ReferenceType="HasEncoding"
                       IsForward="false">ns=1;i=3</Reference>
        </References>
    </UAObjectType>
```

rempAut, 2024-02-16:
Upon further reading, it looks this was a bug and #1550 fixes this.