FreeOpcUa/opcua-asyncio #1529: Browsing for HasTypeDefinition references is broken

kevinherron, 2023-12-17:
Browsing for HasTypeDefinition references is broken.

When specifying a NodeClassMask of only ObjectType or VariableType, no results are returned.

If the NodeClassMask is set to 0 (all NodeClasses should be returned), then 1 result is returned, however it incorrectly identifies the NodeClass as `NodeClass.DataType`. (I suspect this is the cause of the 0 length result when a mask is supplied).

This code written against Eclipse Milo dev/1.0 branch reproduces, though you should be able to reproduce from any OPC UA SDK: https://gist.github.com/kevinherron/f85ce3b0eebe84008e59a5192f6b404d

Results:
```
Browsing with NodeClass mask: 24
  No references found
Browsing with NodeClass mask: 0
  Reference[0]:
    NodeId: ExpandedNodeId{ns=0, id=2138, serverIndex=0}
    BrowseName: QualifiedName{name=ServerStatusType, namespaceIndex=0}
    DisplayName: LocalizedText{text=ServerStatusType, locale=null}
    NodeClass: DataType
    TypeDefinition: ExpandedNodeId{ns=0, id=0, serverIndex=0}
```

Wireshark capture: [browse_has_type_definition.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/13696531/browse_has_type_definition.zip)

kevinherron, 2024-03-01:
I dug into this a little in hopes the additional information might nudge a maintainer to attempt to fix this.

For Nodes belonging to the standard OPC UA address space (ns0), the problem can be traced back to here: https://github.com/FreeOpcUa/opcua-asyncio/blob/d4d4c84366dea6bc0ede632fff0821b00f85d8eb/schemas/generate_address_space.py#L345-L359

where the target NodeClass of every ReferenceDescription created is hardcoded as `NodeClass.DataType`.

In custom address spaces it can be traced to here: https://github.com/FreeOpcUa/opcua-asyncio/blob/d4d4c84366dea6bc0ede632fff0821b00f85d8eb/asyncua/server/address_space.py#L382-L389

where again, the target NodeClass is hardcoded as `NodeClass.DataType`.

BruceGitHub2015, 2024-05-18:
It is just a typo. Change DataType to ObjectType will fix it.

kevinherron, 2024-05-18:
> It is just a typo. Change DataType to ObjectType will fix it.

No, it won't. The target of a HasTypeDefinition reference can be either `ObjectType` _or_ `VariableType`, depending on whether the source is an `Object` or ` Variable`.

BruceGitHub2015, 2024-05-20:
> > It is just a typo. Change DataType to ObjectType will fix it.
> 
> No, it won't. The target NodeClass of a HasTypeDefinition reference can be either `ObjectType` _or_ `VariableType`, depending on whether the source is an `Object` or `Variable`.

I was brought here from this post https://forum.inductiveautomation.com/t/error-calling-opc-ua-method/83234
I can verify that by changing the DataType to ObjectType, the error goes away.

kevinherron, 2024-05-20:
Yes, understood. It would fix that particular issue.

It's not a general fix, though, because it just changes the hardcoded value from one incorrect NodeClass to another.

Whether that's acceptable or not is up to the maintainers of this project, I guess.

BruceGitHub2015, 2024-05-20:
Yes you are right. As other OPC UA clients can handle the original mistake without an issue, I guess it also has something to do with how a client tolerates such problems. As everything on the server is an Object, I thought maybe changing it to a more general ObjectType could make the mistake more tolerable. I barely know anything about ignition so I couldn't really explain how that change affects it.

LaurenceGA, 2024-10-21:
Any chance there will be traction on this? Also running into this issue... :\

kevinherron, 2024-10-22:
> Any chance there will be traction on this? Also running into this issue... :\

If you are a Milo user running into this I’ve made a change in the `dev/1.0` branch to work around this bug. If not, good luck 🙃

LaurenceGA, 2024-10-22:
I see... Thanks for the pointer. I guess I'll go check out what the node-opcua people are up to or something then :)

AndreasHeine, 2024-10-23:
i can confirm this issue!

https://reference.opcfoundation.org/Core/Part3/v104/docs/7.13

> 7.13 HasTypeDefinition ReferenceType
> 
> Errata [exists](https://opcfoundation.org/documents/10000-79/) for this version of the document.
> 
> The HasTypeDefinition ReferenceType is a concrete ReferenceType and can be used directly. It is a subtype of NonHierarchicalReferences.
> 
> The semantic of this [ReferenceType](https://reference.opcfoundation.org/search/13?t=ReferenceType) is to bind an [Object](https://reference.opcfoundation.org/search/13?t=Object) or [Variable](https://reference.opcfoundation.org/search/13?t=Variable) to its [ObjectType ](https://reference.opcfoundation.org/search/13?t=ObjectType)or [VariableType](https://reference.opcfoundation.org/search/13?t=VariableType), respectively[.](https://reference.opcfoundation.org/search/13?t=.) The relationships between types and instances are described in [4.5](https://reference.opcfoundation.org/Core/Part3/v104/docs/?r=_Ref115168300).
> 
> The [SourceNode](https://reference.opcfoundation.org/search/13?t=SourceNode) of this [ReferenceType](https://reference.opcfoundation.org/search/13?t=ReferenceType) shall be an [Object](https://reference.opcfoundation.org/search/13?t=Object) or [Variable](https://reference.opcfoundation.org/search/13?t=Variable). If the [SourceNode](https://reference.opcfoundation.org/search/13?t=SourceNode) is an [Object](https://reference.opcfoundation.org/search/13?t=Object), then the [TargetNode](https://reference.opcfoundation.org/search/13?t=TargetNode) shall be an [ObjectType](https://reference.opcfoundation.org/search/13?t=ObjectType); if the [SourceNode](https://reference.opcfoundation.org/search/13?t=SourceNode) is a [Variable](https://reference.opcfoundation.org/search/13?t=Variable), then the [TargetNode](https://reference.opcfoundation.org/search/13?t=TargetNode) shall be a [VariableType](https://reference.opcfoundation.org/search/13?t=VariableType).
> 
> Each [Variable](https://reference.opcfoundation.org/search/13?t=Variable) and each [Object](https://reference.opcfoundation.org/search/13?t=Object) shall be the [SourceNode](https://reference.opcfoundation.org/search/13?t=SourceNode) of exactly one [HasTypeDefinition](https://reference.opcfoundation.org/search/13?t=HasTypeDefinition) [Reference](https://reference.opcfoundation.org/search/13?t=Reference).

according to the spec the property has to be writen case sensitive!

https://github.com/FreeOpcUa/opcua-asyncio/blob/d4d4c84366dea6bc0ede632fff0821b00f85d8eb/asyncua/server/address_space.py#L382-L389

we need to add a lookup in "_add_type_definition" -> "self.aspace.read_attribute_value(nodeid: ua.NodeId, attr: ua.AttributeIds)" looking for the sourcenode and check its TypeDefinition

oroulet, 2024-10-23:
in that particular function I see that item is of type AddNodeItem and it has an attirbute NodeClass_ . Can't I simply use that one?

AndreasHeine, 2024-10-23:
if it reflects the behavior mentioned above, yes!

kevinherron, 2024-10-23:
It looks to me that `item` in this case represents the Node a type definition is being added to, not the type definition Node itself. I don't think it's going to be the correct NodeClass.

cziebuhr, 2025-03-07:
Had the same issue when trying to read events with AspenTech Inmation and created a fix for it.

PR #1794 (2025-03-09): Fix Nodeclass in ReferenceDescriptions of UA-Nodeset
Fixes #1529.

Add references after all nodes has been processed, when we know the respective NodeClass of the TargetNode. Reused schema version 1.05.01, to not introduce additional changes to the address space.

I also fixed the generation of binary_address_space.pickle, which probably wasn't up-to-date before, unless the generator has been run twice. Not sure if this file is needed at all, at least it's not used by asyncua itself.
closed 2025-03-09 by oroulet