FreeOpcUa/opcua-asyncio #1537: Usage of ua.QueryFirstRequest() not possible

constructionrobotic, 2023-12-21:
When using ua.QueryFirstRequest() as a client, the server response is that this service is not supported. 
Is there any way to fix this or extend the server to handle a query service as mentioned in OPC UA Part 4?

Client-Code:

` 
                data2 = await self.client.uaclient.protocol.send_request(
                    ua.QueryFirstRequest(Parameters=ua.QueryFirstParameters(

                            NodeTypes=[ua.NodeTypeDescription(
                                TypeDefinitionNode=ua.ExpandedNodeId(58,0,ua.NodeIdType.TwoByte, "http://opcfoundation.org/UA/", 0),
                                IncludeSubTypes=True,
                                DataToReturn=[ua.QueryDataDescription(
                                    AttributeId=3,
                                )])],
                            MaxDataSetsToReturn=10,
                            MaxReferencesToReturn=10
                        )
                    )
                )
                response2 = struct_from_binary(ua.QueryFirstResponse, data2)
                print(response2.Parameters)
                
`

Client-Logger:

`
asyncua.ua.uaerrors._auto.BadServiceUnsupported: The server does not support the requested service.(BadServiceUnsupported)
`

Server-Logger:
`
WARNING:asyncua.server.uaprocessor:Unknown message received NodeId(Identifier=615, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) (User(role=<UserRole.User: 3>, name=None))
ERROR:asyncua.server.uaprocessor:sending service fault response: The server does not support the requested service. (BadServiceUnsupported)
`

schroeder-, 2023-12-21:
Asyncua doens't support the QueryRequest on the server side. Feel free to implement it and make a pr.

schroeder-, 2023-12-21:
Also a lot of servers doesn't support views, this is a less common used feature of opc ua.