FreeOpcUa/opcua-asyncio #1538: how to discover local server from remote client?

xibeisiber, 2023-12-21:
Hi,

Since OPCUA servers are deploy on local devices, how can an OPCUA client from cloud/internet discover the servers?
Is there any example or demo on this?

Thanks!

schroeder-, 2023-12-21:
There is OPC UA [GDS](https://opcfoundation.org/developer-tools/documents/view/169), i think it is not working with the asyncua library out of the box. We should have all the building blocks, but you have put them together, from the spec.

xibeisiber, 2023-12-22:
Thanks for your reply. I will look into the GDS

xibeisiber, 2023-12-22:
hi, I understand the GDS is helping remote client to find the endpoints of local servers. These endpoints includes local ip address. But if the device hosting the local servers does not have a public domain, how can remote client  really access these servers? Does the local device need something like "intranet penetration".

And is there an open source GDS avaiable?

schroeder-, 2023-12-22:
you need to expose it to the internet, like a webserver. 

As alternative there is reverse connection. Now your server tries to connect to a client. The downside is, that the server must know the client address. Also this is currently not supported by asyncua. 

A open source implementation is this https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/master/Samples/GDS/README.md