FreeOpcUa/opcua-asyncio #1567: RuntimeError: Unknown datatype for field: StructureField

lueckemeier, 2024-02-07:
When the data types are loaded, the following error appears:

`RuntimeError: Unknown datatype for field: StructureField(Name='Life', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier='anontype4', NamespaceIndex=28, NodeIdType=<NodeIdType.String: 3>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:Tl_V14_t, please report`

The following method is used:
`await client.load_data_type_definitions()`

As it stands, the package cannot instantiate structure elements that are not present in the address space of the structure.
However, according to the OPC UA Spec. this is permitted, the server does not have to provide "HasSubtype References", but an "inverse Reference".
This means that the data type can be determined when reading the data.

However, the package still seems to be strongly orientated towards the old "Dictionary Concept" and also attempts to browse the data types provided according to 1.04 OPC UA Spec. at startup.
The advantage of the "new" concept is that not all data types provided by the server have to be browsed at startup, which has a positive effect on the start of the application.

The client would simply have to skip data types that it cannot browse at startup and should not abort.

Python-Version: 3.11.0
opcua-asyncio Version: 1.0.6

minimanue73, 2024-04-11:
Hello, I had the same error 

> "RuntimeError: Unknown datatype for field: StructureField(Name='CommunicationLinks', Description=LocalizedText(Locale=None, Text=''), DataType=NodeId(Identifier=3007, NamespaceIndex=4, NodeIdType=<NodeIdType.Numeric: 2>), ValueRank=-1, ArrayDimensions=None, MaxStringLength=0, IsOptional=False) in structure:ConnectionEndpointConfigurationDataType, please report" 

when I've tried to import:

> Opc.Ua.Fx.NodeSet2.xml

The xml file has been downloaded from the OPC-UA Foundation. 

I've tried to parse it using the function "self.server.import_xml". Is there an other way to import it ?

huisman2501, 2025-09-10:
Same problem here. I've updated the nodeset to the latest commit to exclude potential fixes from the nodeset side. 

I would like to import the XML to communicate with an Fx based server