FreeOpcUa/opcua-asyncio #1583: Connect to Kepware KEPServerEX 6.15 via asyncua

shamyjun22, 2024-02-18:
Hi there,
I am running an asyncua v1.0.6 client to connect to my Kepware KEPServerEX 6.15 installed in my local machine. 
I can run and read values successfully with settings:
**OPC UA Server Interface > Allow anonymous login set **:  Yes
**Server Endpoints > Securirty Policies**:  None

Now, I wanted change the configuration to these to implement certificate and key:
**OPC UA Server Interface > Allow anonymous login set **:  No
**Server Endpoints > Securirty Policies**:  Basic256Sha256, Sign and Encrypt

So, I have an ssl.conf to generate certificate and key with openssl:
**ssl.conf**:
[ req ]
default_bits = 2048
default_md = sha256
distinguished_name = subject
req_extensions = req_ext
x509_extensions = req_ext
string_mask = utf8only
prompt = no

[ req_ext ]
basicConstraints = CA:FALSE
nsCertType = client, server
keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage= serverAuth, clientAuth
nsComment = "OpenSSL Generated Certificat"
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
subjectAltName = URI:urn:PC1-LT16:Kepware.KEPServerEX.V6:UA%20Server,IP: 192.168.100.139

[ subject ]
countryName = PH
stateOrProvinceName = PH
localityName = PH
organizationName = ShammyJun
commonName = PythonOpcUaServer

**Run Openssl command:**
1. Generate key: "openssl genrsa -out key.pem 2048"
2. Generates a x509v3 certificate: "openssl req -x509 -days 365 -new -out cert.pem -key key.pem -config ssl.conf"

**Source code:** client.py:
`import asyncio

from asyncua import Client
from datetime import datetime

async def main():
    async with Client("opc.tcp://192.168.100.139:49320",) as client:
        await client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")
        client.set_user("Administrator")
        client.set_password("admin1234!@#$")
        client.application_uri = "urn:PC1-LT16:Kepware.KEPServerEX.V6:UA%20Server"

        dateNow = datetime.now()
        volume = await client.get_node('ns=2;s=Modbus.GATEWAY.Volume').read_value()
        power = await client.get_node('ns=2;s=Modbus.GATEWAY.Power').read_value()
        temperature = await client.get_node('ns=2;s=Modbus.GATEWAY.Temperature').read_value()
        watts = await client.get_node('ns=2;s=Modbus.GATEWAY.Watts').read_value()

        print([dateNow.strftime('%Y-%m-%d %H:%M:%S'),str(volume),str(power),str(temperature),str(watts)])

if __name__ == "__main__":
    asyncio.run(main())`
 
**Upon running, the error was:**
ServiceFault (BadSecurityChecksFailed, diagnostics: DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=None, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None)) from server received  in response to CreateSessionRequest
Traceback (most recent call last):
  File "C:\opc\client.py", line 22, in <module>
    asyncio.run(main())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 654, in run_until_complete   
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:opc\client.py", line 7, in main
    async with Client("opc.tcp://192.168.100.139:49320",) as client:
  File "C:\Python3.11\Lib\site-packages\asyncua\client\client.py", line 94, in __aenter__
    await self.connect()
  File "C:\Python3.11\Lib\site-packages\asyncua\client\client.py", line 310, in connect
    await self.create_session()
  File "C:\Python3.11\Lib\site-packages\asyncua\client\client.py", line 489, in create_session
    response = await self.uaclient.create_session(params)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python3.11\Lib\site-packages\asyncua\client\ua_client.py", line 335, in create_session
    data = await self.protocol.send_request(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python3.11\Lib\site-packages\asyncua\client\ua_client.py", line 172, in send_request
    self.check_answer(data, f" in response to {request.__class__.__name__}")
  File "C:\Python3.11\Lib\site-packages\asyncua\client\ua_client.py", line 181, in check_answer
    hdr.ServiceResult.check()
  File "C:\Python3.11\Lib\site-packages\asyncua\ua\uatypes.py", line 375, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadSecurityChecksFailed: An error occurred verifying security.(BadSecurityChecksFailed)

Here the additional information:
**Python:** v3.11.8
**asyncya:** v1.0.6

**Kepware KEPServerEX version:**
[image]

Hopefully, I could get help and fix the error. Or, if I missed or make any mistake, you can provide any inputs. 
Thanks in advance

schroeder-, 2024-02-18:
With the following line you do connect to the server.
`async with Client("opc.tcp://192.168.100.139:49320") as client:`

Rewrite the code like this:

```
async def main():
      # configure client
      client = Client("opc.tcp://192.168.100.139:49320")
      client.set_user("Administrator")
      client.set_password("admin1234!@#$")
      client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.der,key.pem")
      client.application_uri = "urn:PC16-LT16:Kepware.KEPServerEX.V6:UA%20Server"
      # connect to the server
      async with client:
          # connected
           .....
```

closed 2024-02-18 by schroeder-

shamyjun22, 2024-02-18:
Hi @schroeder, thanks.