FreeOpcUa/opcua-asyncio #1584: The following references could not be imported and are probably broken.

varunJainV, 2024-02-20:
### Could not be imported!

---
 **self.server = Server()
 await self.server.init()
 self.server.set_endpoint(self.endpoint)
 await self.server.import_xml(self.xmlfile)**

In import_xml method I am getting following error: 
**The following references could not be imported and are probably broken.**
[image]

schroeder-, 2024-02-20:
Can you post a complete log, not an incomplete screenshot?

Also can you post the xml file if possible? Otherwise can you post a log with log level Debug?

varunJainV, 2024-02-20:
Thank You for your response. 
Here  I am sharing the complete screen shot and xml file.
 
[image]
[image]

Link of the XML file [https://reference.opcfoundation.org/api/nodesets/61](url)

schroeder-, 2024-02-20:
Do you load the required nodesets before? http://opcfoundation.org/UA/DI/ and http://opcfoundation.org/UA/PlasticsRubber/GeneralTypes/ ?

varunJainV, 2024-02-20:
Iam using the following to load nodesets:
[image]

schroeder-, 2024-02-20:
Do you even read my responeses....

closed 2024-02-20 by schroeder-

varunJainV, 2024-02-20:
Can you please confirm, How to load them?