FreeOpcUa/opcua-asyncio #1589: 1.0.7

AiyionPrime, 2024-02-22:
It's been a little over two month since the last release.

@oroulet and @schroeder- Are there PRs you'd see as blocking a potential next release?

I'd start testing the current master-branch as release candidate until you answered.

AiyionPrime, 2024-02-28:
Lacking a release candidate, I've tested master at 50566320c33e1df065ca2ffef99046f9f2f7bff3.
Works as expected; at least for me.

@oroulet @schroeder- If there's anything I can help moving this forward just let me know.

As there were several features and not only bugfixes since the last release
`1.1.0` appears to be more appropriate than `1.0.7`.

This project does not do semver, does it?
(Small) breaking changes in the public do not necessarily lead to a new major version?

oroulet, 2024-02-28:
done

closed 2024-02-29 by AiyionPrime