FreeOpcUa/opcua-asyncio #1612: Stuck in obtaining the read_data_type() of node. Related to asyncio feature.

DiogoFran, 2024-04-09:
**Describe the bug** <br />
The process of acquiring a read_data_type() of node stayed stuck and i don't know why, i don't know if it's my mistake.

**To Reproduce**<br />
I have a module python (x.py) that calls the object My_OPCUA_Client, as follow:
[image]
My Object My_OPCUA_Client.
[image]
After creating the OPCUA client instance i call from x.py the function getNodeId() which is async, as shown above:
[image]

After that, the python interpreter stayed stuck. The result that i obtained is:
[image]
So, it starts the awaitable coroutine read_data_type() but doesn't return.

**Expected behavior**<br />
Obtaining the data type of node asynchronously

**Version**<br />
Python-Version:3.11.5
asyncua==1.0.4

Thanks for any help.

oroulet, 2024-04-17:
you need to do a bit more asyncio training. you cannot use asyncio.run that way

closed 2024-04-17 by oroulet

DiogoFran, 2024-04-17:
Hello, could you give me some tip? 
Thank you

schroeder-, 2024-04-17:
1. First dont post screenshoots, always add code as text. Also log outputs. It's hard to read and understand screenshoots.
2. read about asyncio
3. Here a incomplete example to use asyncio for your code:
Remove asyncio.run(create_client()) from My_OPC_UA_Client __init__

```
async def run_client():
    client = My_OPC_UA_Client():
    await client.create_client()
    await client.get_node_ide("id")

if __name__ == "__main__":
    asyncio.run(run_client())
   
```

DiogoFran, 2024-04-17:
Sorry for the screenshots. 
I have read about that, the option mentioned only works if i want to run the script as the initial program entry point which doesn't happen.  
I think the flow of my code, finishes the first asyncio.run() (when the object is created) and after goes to the asyncio.run() to execute the async function "getNodeId". I am not understanding the error.

oroulet, 2024-04-17:
you need to have only one asyncio loop for the entire session (roughly). in a big application the the loop should be running already. Sounds like you in reality want to write sync code. Use the sync wrapper then

DinoFra, 2024-05-06:
Hello again, i could not find the answer.
I verified if the application already had a loop, but it didn't.   I just want to initialize an object with an already configured client. After that, i want to call the object to pick up nodes when i need it.
What's the wrapping you're talking about?