FreeOpcUa/opcua-asyncio #1614: Exception in subscription loop: struct.error: required argument is not an integer

magnus-bakke, 2024-04-12:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/opcua-asyncio/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/opcua-asyncio

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
We are running asyncua in a Docker container using an image based on `python:3.10-slim-bullseye`.

Because the person who developed and monitored this solution has left the company, I can't say when it was last working. All I know is that:

- We have a bunch of IoT devices to which we can connect via TCP and retrieve data.
- We have a Python repository used as an interface for reading this data from the devices (which I'll call the "ethernet adapter").
- We have a Python repository (the "OPC UA repo") that uses asyncua to populate nodes/values. This repo has a loop that uses the ethernet adapter to read device data.
- To release the OPC UA repo, we build its Docker image and push it to an image repository in the cloud.
- Finally, a "master repo" pulls the image from the cloud, along with several other images, and runs it using Docker Compose.

After launching the "master" application, which coordinates images (including the OPC UA repo that uses asyncua), I look at its logs and see this error and traceback for the OPC UA container:

```
Exception in subscription loop
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/asyncua/server/internal_subscription.py", line 88, in _subscription_loop
    await self.publish_results()
  File "/usr/local/lib/python3.10/site-packages/asyncua/server/internal_subscription.py", line 135, in publish_results
    await self.pub_result_callback(result, requestdata)
  File "/usr/local/lib/python3.10/site-packages/asyncua/server/uaprocessor.py", line 94, in forward_publish_response
    self.send_response(requestdata.requesthdr.RequestHandle, requestdata.seqhdr, response)
  File "/usr/local/lib/python3.10/site-packages/asyncua/server/uaprocessor.py", line 49, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 336, in struct_to_binary
    return serializer(obj)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 327, in <genexpr>
    else serializer(obj.__dict__[name])
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 327, in <genexpr>
    else serializer(obj.__dict__[name])
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 327, in <genexpr>
    else serializer(obj.__dict__[name])
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 401, in serialize
    return data_size + b''.join(type_serializer(el) for el in val)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 401, in <genexpr>
    return data_size + b''.join(type_serializer(el) for el in val)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 569, in extensionobject_to_binary
    body = struct_to_binary(obj)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 336, in struct_to_binary
    return serializer(obj)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 327, in <genexpr>
    else serializer(obj.__dict__[name])
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 401, in serialize
    return data_size + b''.join(type_serializer(el) for el in val)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 401, in <genexpr>
    return data_size + b''.join(type_serializer(el) for el in val)
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 325, in serialize
    return b''.join(
  File "/usr/local/lib/python3.10/site-packages/asyncua/ua/ua_binary.py", line 327, in <genexpr>
[... 139 lines cut ...]
    return master_data
```

**Expected behavior**<br />
A clear and concise description of what you expected to happen.

When viewed using UaExpert, these columns should be populated as there are no polling errors:

[image]

(and many more)

**Screenshots**<br />
If applicable, add screenshots to help explain your problem. <br />
You can post Screenshots directly to github, please don't use 3rd Party webpages

See also the expected behavior.

[image]

(Notice that it continues to fail to write a bunch of None values. These None errors are expected.)

**Version**<br />
Python-Version:<br />3.10
opcua-asyncio Version (e.g. master branch, 0.9): 1.1.0

oroulet, 2024-04-17:
also the write has been done server side somewhere (we do not do type checking at that point, it is too slow), but the parsing only happens when a client subscribe to it and the server try to send the data

magnus-bakke, 2024-04-17:
> your code is somewhere trying to write a struct to the ua server that cannot be parsed. Looks like something that is declared as an int or where an int is expected is not an int...

You're right, a float was declared as an int. Would it not be a simple thing to print the node ID and variant type of the offender when it's parsed?

oroulet, 2024-04-17:
Usually it does. Looks like you hit a place where we do not.... MR welcome

closed 2026-08-15 by oroulet