FreeOpcUa/opcua-asyncio #1639: BadIdentityTokenInvalid when connecting to open62541 based server

lauroasr, 2024-05-08:
Hi, I'm having a problem connecting the asyncua client to a third-party OPC UA server from a company. It looks like this company uses a open62541 based server (as specified by the `ApplicationName` from the logs).

**Behavior**
The access works as the following: The client generates the certificate and uses that to connect to the server. A popup appears on the third-party system to accept the certificate. Once accepted, I try connecting again and get the following error: 
```bash
asyncua.ua.uaerrors._auto.BadIdentityTokenInvalid: The user identity token is not valid.(BadIdentityTokenInvalid)
```

**Client**
My client is mostly based on the certificate authentication example found in the repository. 

```python
import asyncio
import logging
import sys
import socket
from pathlib import Path
from cryptography.x509.oid import ExtendedKeyUsageOID
sys.path.insert(0, "..")
from asyncua import Client
from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256
from asyncua.crypto.cert_gen import setup_self_signed_certificate
from asyncua.crypto.validator import CertificateValidator, CertificateValidatorOptions
from asyncua.crypto.truststore import TrustStore
from asyncua import ua

logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger(__name__)

cert_base = Path(__file__).parent
cert = Path(cert_base / f"certs/peer-certificate-2.der")
private_key = Path(cert_base / f"certs/peer-private-key-2.pem")

async def task(loop):
    host_name = socket.gethostname()
    client_app_uri = f"urn:{host_name}:myapp"
    url = "opc.tcp://192.168.69.152:16664/"

    await setup_self_signed_certificate(private_key,
                                        cert,
                                        client_app_uri,
                                        host_name,
                                        [ExtendedKeyUsageOID.CLIENT_AUTH],
                                        {
                                            'countryName': 'CN',
                                            'stateOrProvinceName': 'AState',
                                            'localityName': 'Foo',
                                            'organizationName': "Bar Ltd",
                                        })
    client = Client(url=url)
    client.application_uri = client_app_uri
    await client.set_security(
        SecurityPolicyBasic256Sha256,
        certificate=str(cert),
        private_key=str(private_key)
    )

    async with client:
        print('Connection worked!')
        await asyncio.sleep(2)

def main():
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(task(loop))
    loop.close()

if __name__ == "__main__":
    main()
```

**Screenshots**
I have a UAExpert client that connects successfully even when I use **the same certificate** generated by the python client. This is how the configuration is there:

[image]

These are the warnings that appear regarding the IP address not matching what is on the certificate, but once I hit `Ignore` the connection is successful:

[image]
[image]

**Version**
Python version: 3.12.1
asyncua version: 1.1.0

I tried some things, including some old issues posted here (e.g. #742 and #227) but it looks like this case is different. Any help would be much appreciated!

lauroasr, 2024-05-08:
This is the output with logging level set to `DEBUG`

<details>
  <summary><b>Output</b></summary>
  PS C:\Users\lauroasr\Documents\asyncua-test&gt; python .\asyncua-client.py
C:\Users\lauroasr\AppData\Roaming\Python\Python312-32\site-packages\OpenSSL\_util.py:6: UserWarning: You are using cryptography on a 32-bit Python on a 64-bit Windows Operating System. Cryptography will be significantly faster if you switch to using a 64-bit Python.
  from cryptography.hazmat.bindings.openssl.binding import Binding
C:\Users\lauroasr\Documents\asyncua-test\asyncua-client.py:53: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
DEBUG:asyncio:Using proactor: IocpProactor
WARNING:asyncio:Executing &lt;Task pending name='Task-1' coro=&lt;task() running at C:\Users\lauroasr\Documents\asyncua-test\asyncua-client.py:28&gt; wait_for=&lt;Future pending cb=[_chain_future.&lt;locals&gt;._call_check_cancel() at C:\Program Files (x86)\Python312-32\Lib\asyncio\futures.py:387, Task.task_wakeup()] created at C:\Program Files (x86)\Python312-32\Lib\asyncio\base_events.py:447&gt; cb=[_run_until_complete_cb() at C:\Program Files (x86)\Python312-32\Lib\asyncio\base_events.py:180] created at C:\Program Files (x86)\Python312-32\Lib\asyncio\tasks.py:695&gt; took 0.328 seconds
C:\Users\lauroasr\AppData\Roaming\Python\Python312-32\site-packages\asyncua\crypto\uacrypto.py:334: CryptographyDeprecationWarning: Properties that return a na&iuml;ve datetime object have been deprecated. Please switch to not_valid_after_utc.
  if cert.not_valid_after &lt; now:
C:\Users\lauroasr\AppData\Roaming\Python\Python312-32\site-packages\asyncua\crypto\uacrypto.py:337: CryptographyDeprecationWarning: Properties that return a na&iuml;ve datetime object have been deprecated. Please switch to not_valid_before_utc.
  if cert.not_valid_before &gt; now:
INFO:asyncua.client.ua_client.UaClient:opening connection
DEBUG:asyncio:&lt;asyncio.TransportSocket fd=904, family=2, type=1, proto=6, laddr=('10.10.10.50', 53235), raddr=('192.168.69.152', 16664)&gt; connected to 192.168.69.152:16664: (&lt;_ProactorSocketTransport fd=904 read=&lt;_OverlappedFuture pending cb=[_ProactorReadPipeTransport._loop_reading()] created at C:\Program Files (x86)\Python312-32\Lib\asyncio\windows_events.py:493&gt;&gt;, &lt;asyncua.client.ua_client.UASocketProtocol object at 0x05054AF8&gt;)
INFO:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=131072, max_send_buffer=131072, max_chunk_count=32768, max_message_size=1073741824)
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: OpenSecureChannelRequest(TypeId=FourByteNodeId(Identifier=446, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.FourByte: 1&gt;), RequestHeader_=RequestHeader(AuthenticationToken=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.TwoByte: 0&gt;), Timestamp=datetime.datetime(2024, 5, 8, 14, 23, 22, 953921), RequestHandle=1, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=1000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.TwoByte: 0&gt;), Body=None)), Parameters=OpenSecureChannelParameters(ClientProtocolVersion=0, RequestType=&lt;SecurityTokenRequestType.Issue: 0&gt;, SecurityMode=&lt;MessageSecurityMode.None_: 1&gt;, ClientNonce=b'', RequestedLifetime=3600000))
DEBUG:asyncua.client.ua_client.UaClient:get_endpoint
DEBUG:asyncua.client.ua_client.UASocketProtocol:Sending: GetEndpointsRequest(TypeId=FourByteNodeId(Identifier=428, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.FourByte: 1&gt;), RequestHeader_=RequestHeader(AuthenticationToken=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.TwoByte: 0&gt;), Timestamp=datetime.datetime(2024, 5, 8, 14, 23, 23, 110510), RequestHandle=2, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=4000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.TwoByte: 0&gt;), Body=None)), Parameters=GetEndpointsParameters(EndpointUrl='opc.tcp://192.168.69.152:16664/', LocaleIds=[], ProfileUris=[]))
DEBUG:asyncua.client.ua_client.UASocketProtocol:We did not receive enough data from server. Need 5812 got 1310
DEBUG:asyncua.client.ua_client.UASocketProtocol:We did not receive enough data from server. Need 5812 got 5276
DEBUG:asyncua.client.ua_client.UaClient:GetEndpointsResponse(TypeId=NodeId(Identifier=431, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.FourByte: 1&gt;), ResponseHeader_=ResponseHeader(Timestamp=datetime.datetime(2024, 5, 8, 14, 23, 23, 180766), RequestHandle=2, ServiceResult=StatusCode(value=0), ServiceDiagnostics=DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=None, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None), StringTable=None, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=&lt;NodeIdType.TwoByte: 0&gt;), Body=None)), Endpoints=[EndpointDescription(EndpointUrl='opc.tcp://192.168.69.152:16664/', Server=ApplicationDescription(ApplicationUri='urn:open62541.server.application', ProductUri='http://open62541.org', ApplicationName=LocalizedText(Locale='en', Text='open62541-based OPC UA Application'), ApplicationType_=&lt;ApplicationType.Server: 0&gt;, GatewayServerUri=None, DiscoveryProfileUri=None, DiscoveryUrls=['opc.tcp://192.168.69.152:16664/']), ServerCertificate=b'0\x82\x03\xa10\x82\x02\x89\xa0\x03\x02\x01\x02\ [... cut ...]
  File &quot;C:\Users\lauroasr\AppData\Roaming\Python\Python312-32\site-packages\asyncua\client\ua_client.py&quot;, line 349, in activate_session
    response.ResponseHeader.ServiceResult.check()
  File &quot;C:\Users\lauroasr\AppData\Roaming\Python\Python312-32\site-packages\asyncua\ua\uatypes.py&quot;, line 375, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadIdentityTokenInvalid: The user identity token is not valid.(BadIdentityTokenInvalid)
PS C:\Users\lauroasr\Documents\asyncua-test&gt;
</details>

schroeder-, 2024-05-08:
You mix up security and authentication. You currently only setup the secure layer of opc ua. If you want to use a certificate, as you did in UaExpert, you have to call `load_client_certificate` and `load_private_key` before connecting. Authentication and Security are separated in opc ua.

closed 2024-05-08 by schroeder-

lauroasr, 2024-05-08:
Apologies for my confusion. I swear I tried using these but got a different issue, but now it works! I noticed that I need to set both security and authentication. 

To wrap this up: Is it ok to use the same certificate for both? I noticed they had a problem [here](https://github.com/FreeOpcUa/opcua-asyncio/issues/742#issuecomment-993504182). 

Thanks for the help.