FreeOpcUa/opcua-asyncio #1642: Infinite Read Requests?

imaginaryBuddy, 2024-05-14:
**Describe the bug** <br />
When I intend to only send a read request once, using the synchronous version, it tends to cause an infinite read and I have no idea how to fix it. 

**To Reproduce**<br />

```
from asyncua.sync import Client, sync_uaclient_method, sync_async_client_method
from asyncua.client import Client as AsyncClient
import time

try:
        client = Client(<ip>)
        client.connect() 
        node_to_read = [client.get_node("ns=2;i=2")]
        print(node_to_read)
        read = sync_async_client_method(AsyncClient.read_values)(client)
        read(nodes=node_to_read)
        time.sleep(10)
        client.disconnect()
except:
        client.disconnect()
        raise 
```

**Expected behavior**<br />
I expect only 1 packet of read request sent from Client 

**Screenshots**<br />
This is the Wireshark capture after sending 1 read request, and setting a time.sleep(10) before disconnecting. 
[image]

**Version**<br />
Python-Version: 3.12.1
opcua-asyncio Version: 1.1.0

schroeder-, 2024-05-14:
Does this work?

```
try:
        client = Client(<ip>)
        client.connect() 
        node_to_read = [client.get_node("ns=2;i=2")]
        print(node_to_read)
        vals = client.read_values(node_to_read)
        print(vals)
        time.sleep(10)
        client.disconnect()
except:
        client.disconnect()
        raise 
```

imaginaryBuddy, 2024-05-15:
@schroeder- It doesn't work. There's still infinite read requests being sent

schroeder-, 2024-05-15:
Does read a single value work? 

`client.get_node("ns=2;i=2").read_value()`

schroeder-, 2024-05-15:
I checked it this is not a bug.
We have a background task that reads every second to keep the connection alive.
As i read it correct you wait 10 seconds, so 11 readrequests are generated, 1 from your request and 10 from the watchdog.

closed 2024-05-15 by schroeder-

imaginaryBuddy, 2024-05-15:
@schroeder- is there a way to disable this background task ?

schroeder-, 2024-05-15:
Not in the syncclient.

oroulet, 2024-05-15:
you want that task otherwise the client will die at some point. All UA clients do that. you can reduce and change behaviour if you want but you have to read code and override some stuff

imaginaryBuddy, 2024-05-16:
@oroulet, I see, but my current project aims to replay captured packets from Wireshark for debugging purposes, so I do not want any extra read requests as I will ensure that my client stays alive based on the replay. 
could I get some guidance on how to change the behaviour? 

Thank you!

oroulet, 2024-05-16:
Grep code for keepAlive

imaginaryBuddy, 2024-05-16:
I just set watchdog_intervall in the async version of Client to math.inf, and it works for my particular case: 
`self._watchdog_intervall = math.inf`