FreeOpcUa/opcua-asyncio #1673: How to use browsepath to get the value of node id.

codedebugger1, 2024-07-02:
I am have a question on top of this answer:

you can only use nodeids for direct read!
the only thing you can do use the translate browsepath to nodeid service to get the nodeid and then read.

_Originally posted by @AndreasHeine in https://github.com/FreeOpcUa/opcua-asyncio/discussions/1228#discussioncomment-5084140_

Basically our need is how we can read and write the values via browse name.
How to use the browse path in that case.

AndreasHeine, 2024-07-02:
https://github.com/FreeOpcUa/opcua-asyncio/blob/cf27f8cf19e3081317f871550af89c1af28e18df/asyncua/client/client.py#L957

https://reference.opcfoundation.org/Core/Part4/v105/docs/A.2

rel path: "/0:Root/0:Objects/0:Server" (you need to use Browsenames!!! "<NamespaceIndex>:<Name>")

codedebugger1, 2024-07-02:
Thanks for your response.

I have tried with below code and couldn't succeded.
I have tried with relative path only.

     async def main():
     client = Client(url=address, timeout=4)

    try:
        await client.connect()
        print("Success")
    except Exception as e:
        print(e)
        return

    paths = await client.translate_browsepaths(starting_node="ns=1;i=100", relative_paths="/0:Root/0:Objects/1:MyServer/1:StatusWord")
    print("paths", paths)

    try:
        await client.disconnect()
    except Exception as e:
        print(e)
        return

     if __name__ == "__main__":
          asyncio.run(main())

We wanted something like this (below is the example of the reading value from node id, we wanted similar with reading value from browsepath ):

        client.connect()
        print("Connect Success")
        node_obj= client.get_node("ns=1;i=100")
        node_value = node_obj.get_value()
        print("node_value", node_value)

AndreasHeine, 2024-07-02:
> Thanks for your response.
>
> I have tried with below code and couldn't succeded. I have tried with relative path only.
>
> ```
>  async def main():
>  client = Client(url=address, timeout=4)
>
> try:
>     await [...]

sorry in opc ua it to seperate services means always two seperate requests...

not sure if root and object belongs to a "relative" path!? i used it just to show the syntax...

AndreasHeine, 2024-07-02:
its usually from the startnode the path to the node you want to get

guess thos is your rel path: ´/1:MyServer/1:StatusWord´

codedebugger1, 2024-07-02:
Yes my start node is "ns=1;i=100"

And I have tried with the relative_path as "/1:StatusWord" as below:

            asyncua_obj = asyncuaClient(address)
            asyncua_obj.connect()
            paths = asyncua_obj.translate_browsepaths(starting_node=node_address, relative_paths="/1:StatusWord")
            print("paths", paths)
      

Below is the response I got in the paths:

[image]

AndreasHeine, 2024-07-02:
are you familiar with asyncio?

```python
        asyncua_obj = asyncuaClient(address)
        await asyncua_obj.connect()
        paths = await asyncua_obj.translate_browsepaths(starting_node=node_address, relative_paths="/1:StatusWord")
        print("paths", paths)
```

you need to await the connect and the translate method call...

codedebugger1, 2024-07-03:
Thanks for the information.
Now I have used await for the connect and translate method call... 

    node_address = "ns=1;i=100"

    async def connect():
           asyncua_obj = asyncuaClient(address)
           await asyncua_obj.connect()
           paths = await asyncua_obj.translate_browsepaths(starting_node=node_address, relative_paths="/1:StatusWord")
           print("paths", paths)
           await asyncua_obj.disconnect()

    asyncio.run(connect())

And got this error:

[image]

AndreasHeine, 2024-07-03:
https://github.com/FreeOpcUa/opcua-asyncio/blob/cf27f8cf19e3081317f871550af89c1af28e18df/asyncua/client/client.py#L957

as in the type hint its a ua.NodeId and not a string!

```python
node_address = ua.NodeId.from_string("ns=1;i=100")

async def connect():
       asyncua_obj = asyncuaClient(address)
       await asyncua_obj.connect()
       paths = await asyncua_obj.translate_browsepaths(starting_node=node_address, relative_paths="/1:StatusWord")
       print("paths", paths)
       await asyncua_obj.disconnect()

asyncio.run(connect())
```

codedebugger1, 2024-07-03:
Thanks for the prompt response.

Here is what I have modified the code:

    from asyncua import ua as asyncuaUa
    async def connect():
              asyncua_obj = asyncuaClient(address)
              await asyncua_obj.connect()
              node_address = asyncuaUa.NodeId.from_string("ns=1;i=100")
              paths = await asyncua_obj.translate_browsepaths(starting_node=node_address, relative_paths="/1:StatusWord")
              print("paths", paths)
    await asyncua_obj.disconnect()

    asyncio.run(connect())

And I got this now:

[image]

codedebugger1, 2024-07-03:
Further mode I have tried and gave some prints in the client.py code as below:

[image]

And got below response, All the time, the relative path in the browse path is empty as highligted in below screenshot

[image]

AndreasHeine, 2024-07-03:
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/tests/test_client.py#L163-L175