FreeOpcUa/opcua-asyncio #1693: Unable to load ExtensionObject (struct) BinaryIdBaseDataType from Machine Vision CS.

izonfreak, 2024-08-09:
Hello, 

I was testing the Client with an OPC UA server that implements **OPC 40100-1: Machine Vision - Control, Configuration management, recipe management, result management**

https://reference.opcfoundation.org/MachineVision/v100/docs/

NodeSet -> https://github.com/OPCFoundation/UA-Nodeset/tree/latest/MachineVision

The problem is that when I try to load the server data types  (load_data_types_definitions() ) I get the following error:

```

  File "C:\Users\aaa\mvtec\vvv\EventTest.py", line 42, in main
    await client.load_data_type_definitions()
  File "C:\Users\aaa\AppData\Local\Programs\Python\Python311\Lib\site-packag
es\asyncua\client\client.py", line 864, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overw
rite_existing)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
^^^^^^^^^^^^^^
  File "C:\Users\aaa\AppData\Local\Programs\Python\Python311\Lib\site-packag
es\asyncua\common\structures104.py", line 481, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type, lo
g_fail=log_ex)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
^^^^^^^^^^^^^^
  File "C:\Users\aaa\AppData\Local\Programs\Python\Python311\Lib\site-packag
es\asyncua\common\structures104.py", line 295, in _generate_object
    code = make_structure_code(data_type, name, sdef, log_error=log_fail)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\aaa\AppData\Local\Programs\Python\Python311\Lib\site-packag
es\asyncua\common\structures104.py", line 219, in make_structure_code
    raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{stru
ct_name}, please report")
RuntimeError: Unknown datatype for field: StructureField(Name='Id', Description=
LocalizedText(Locale='', Text='Id is a system-wide unique name for identifying t
he recipe.'), DataType=NodeId(Identifier=3017, NamespaceIndex=2, NodeIdType=<Nod
eIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOp
tional=False) in structure:BinaryIdBaseDataType, please report

Process finished with exit code 1
```
This structure data type BinaryIdBaseDataType is an abstract and is the parent to other data types such as "ConfigurationIdDataType". 

[image]

Since the exception mentions to report it here i am. 

Python-Version: 3.11.4
opcua-asyncio Version 1.1.0

schroeder-, 2024-08-28:
In our tests we load the machine vision cs https://github.com/FreeOpcUa/opcua-asyncio/blob/af1b102b8c4f039a36e70f71f97019e12d2dfce9/tests/test_xml.py#L86 

So this looks like a problem with the server.

izonfreak, 2024-09-02:
Hello @schroeder- , 

The test seems to be for server part. I am trying to connect as a client, and that is where the data type definitions fail.  It seems that the problem comes from having a Structure with a field that is not one of the built-in data types.  

[image]
 
In the case BinaryIdBaseDataType has a field that is of TrimmedString which is a sup-type of string (also define in the same xml file).  Checking the code in structures104.py ( I am no python expert btw) and it seems that it should work right ? specifically :

      elif sfield.DataType in ua.basetype_by_datatype:
            uatype = ua.basetype_by_datatype[sfield.DataType]

TrimmedString should be an alias for "String" so this should return true.. but somehow it does not.  I am correct in my understanding ? Could this be an ordering issue in the XML schema ? 

I did also try to just import the xml file as a client but I get also an strange error "BadUserAccesDenied" and some other warnings. I also could not import it without setting strict mode to false. 

        await client.import_xml('schema/Opc.Ua.MachineVision.NodeSet2.xml', strict_mode=False)

sebastianpfischer, 2025-11-18:
Hello,
we are also impacted by this issue. All Datatypes are stored in the `ua` module. If there was a way to create one abstraction and support namespaces there, it may resolve the problem.

@schroeder- ?