FreeOpcUa/opcua-asyncio #1699: Client.browse_nodes returns no references when a node has children

mng1dev, 2024-08-26:
**Describe the bug** <br />

Hello, I am successfully connecting to my OPC server and would like to discover all the children nodes starting from a specific one.

To reduce network overheads, I want to use BrowseRequests, which allows me to send a batch of NodeIds in a single request, but when I do so, the References list of the BrowseResult is empty.

This problem doesn't occur when I use get_children to retrieve the children of a single node.

**To Reproduce**<br />
```python
root_node = client.get_node('ns=2;s=/my/node/')
print('Children', await root_node.get_children()) # Works! 
browsing = await client.browse_nodes([root_node])
print('Browse', browsing) # References of the BrowseResult is an empty list ...
```

**Expected behavior**<br />

The References list, should contain all the children nodes instead of being empty.

**Version**<br />
Python-Version: 3.10
opcua-asyncio: master branch (1.1.5)

AndreasHeine, 2024-08-26:
whats the referencetype? HasComponent? HasProperty?

AndreasHeine, 2024-08-26:
you can try a "low-level" browse too see:

https://github.com/AndreasHeine/opcua-tutorial/blob/main/client/browse.py#L41-L57

mng1dev, 2024-08-26:
Thanks for your help @AndreasHeine , the ReferenceTypeId I am getting is 35 (Organizes).

AndreasHeine, 2024-08-26:
the thing is on browsing you can specify a lot of option maybe the defaults dont work in browse_nodes...

AndreasHeine, 2024-08-26:
> Thanks for your help @AndreasHeine , with this low-level call, I can see the children nodes.
> 
> The ReferenceTypeId I am getting is 35 (Organizes).

https://github.com/AndreasHeine/opcua-tutorial/blob/70484d292619f9a3fa842b8352b4459c243baa53/client/browse.py#L48C5-L48C42

because is specificaly looks for Organizes references

mng1dev, 2024-08-26:
> because is specificaly looks for Organizes references

By removing this line `bd.ReferenceTypeId = ua.NodeId(35, 0)` I get an empty list of references, so it looks like this parameter has to be configured. In the current implementation, `browse_nodes` doesn't seem to configure that parameter.

AndreasHeine, 2024-08-26:
thats correct!

33 (HierarchicalReferences) + IncludeSubtypes = "All HierarchicalReferences and its Subtypes"

[image]

but be aware that there are many subtypes!

mng1dev, 2024-08-27:
Given that the purpose of my crawling was to "explore" the nodes, I managed to narrow down the list of reference type ids to these:

```python
referencetypes = [
    35 # Organizes
    , 44 # Aggregates
    , 45 # HasSubType
]
```

By taking inspiration from an [OPC browser](https://github.com/node-opcua/opcua-commander/blob/b7b805936e6d7068281da2d3fa7f17522ec34a03/lib/model/model.ts#L540) implemented using `node-opcua`.

I still think the `Client.browse_nodes()` method is not correctly implemented, because no referenceTypeId is specified in that logic, resulting in an empty list of references.

closed 2024-08-28 by mng1dev