FreeOpcUa/opcua-asyncio #1705: Receive BadEncoding error when subscribing to alarms

marc-batto, 2024-09-03:
**Describe the bug** <br />
When following the example (https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client_to_uanet_alarm_conditions.py) to subscribe to alarms and conditions events the client receives a BadEncoding error and disconnects. 

**To Reproduce**<br />

The code I use is a bit more elaborate than this but this is the minimum required to get the error we are seeing:

```python
import asyncio
import os
import sys

from asyncua import Client, ua

URL = os.getenv("OPC_SERVER_ADDRESS")

class SubHandler:
    async def event_notification(self, event):
        print(event)

async def main():
    async with Client(url=URL, ) as client:
        alarmConditionType = client.get_node("ns=0;i=2915")

        # Create subscription for AlarmConditionType
        msclt = SubHandler()
        sub = await client.create_subscription(0, msclt)
        
        handle = await sub.subscribe_alarms_and_conditions(sourcenode=client.nodes.server, evtypes=alarmConditionType) # <-- this call fails with the badencodingerror
        await asyncio.sleep(10)

if __name__ == '__main__':

    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    try:
        loop.run_until_complete(main())
    finally:
        loop.run_until_complete(loop.shutdown_asyncgens())
        loop.close()
```

After running this code the connection to the server is created, the node id to the root node is succesfully retrieved but we see an error message like this from the `subscribe_alarms_and_conditions` call:

```
CRITICAL:asyncua.client.ua_client.UASocketProtocol:Received an error: ErrorMessage(Error=StatusCode(value=2147876864), Reason='')
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
ERROR:asyncua.client.ua_client.UASocketProtocol:Got error status from server: Encoding halted because of invalid data in the objects being serialized.(BadEncodingError)
```

**Expected behavior**<br />
I expect the subhandler to receive alarms when they change their respective values. Or at least the alarm subscription to be created without throwing an error

**Version**<br />
We are working with an Experion HS R520.2
python 3.12.3
opcua-asyncio Version: 1.1.5

marc-batto, 2024-09-04:
Hi @AndreasHeine thanks for the reply, 

It turns out the system we are using does not expose the alarms. From their community center:

```
Summary:
Does Experion PKS R520.1 support OPC UA DA, OPC UA HDA and OPC UA A&E protocol

Question / Problem:
Does Experion PKS R520.1 support OPC UA DA (Data access), OPC UA HDA (History Data Access) and OPC UA  A&E (Alarm & Event) protocol ?

Answer / Solution:
Experion PKS R520.1 supports both OPC UA DA(Data Access) and HDA(History Data Access) but Experion alarms and events (AE) are not currently exposed via OPC UA

OPC UA Data Access support (Support in Experion PKS R511 & R520.1 releases)
OPC UA Historical Access support (New feature in Experion PKS R520.1 release)

```

I will close this issue as it has nothing to do with asyncua probably 👍 

Thanks for this amazing project!

closed 2024-09-04 by marc-batto

AndreasHeine, 2024-09-04:
ok!

usually the OPC UA Server has Profile-Array which contains the supportet Features like A&C
and the Server-Object usually has a referecne thet it emits events and alarms ;)

AndreasHeine, 2024-09-04:
[image]

schroeder-, 2024-09-04:
Did you try to get alarms via UAexpert? If it doesn't work there its not available.